ROBOTIS-GIT/turtlebot3 #64: increase speed for TB3?

idrobot, 2017-06-13:
Hi engineer,

The max speed for Burger is 0.22m/s, and for Waffle is 0.26m/s. Is it possible to increase speed to 0.8m/s - 1.0m/s? for example:
- to use velocity control mode of XM?
- to replace XM430/XL430 to XH series?
- to replace XM430/XL430 to other actuators such as MX-12W, or MX-106?

robotpilot, 2017-06-14:
Hello @idrobot 

- to use velocity control mode of XM?
: All XL/XM/XL series are using velocity control mode.

- to replace XM430/XL430 to XH series?
: The XH430 series also can not operate at the speed you want. 
  - XH430-W210-T / XH430-W210-R
  - XH430-W350-T / XH430-W350-R 
  - XH430-V210
  - XH430-V350   

- to replace XM430/XL430 to other actuators such as MX-12W, or MX-106?
: The MX-12W is faster than the current X series. However, since the MX and X series differ in the fastening structure, they can not be connected directly. It is also not the same as the X series speed control mode.

We received feedback from TurtleBot 3's beta testers that many users wanted faster speeds. Although there is no way to give the user right now, we plan to support this by diversifying the X Series and adjusting the wheel size of TurtleBot3.

Thanks :)

robotpilot, 2017-07-18:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-07-18 by robotpilot