ROBOTIS-GIT/turtlebot3 #66: why i startup r200, ubuntu system in Joule will become very slow.

ncnynl, 2017-07-25:
Today i test r200 camera .  i startup r200_nodelet_default,   several test the ubuntu system always very slow.  
But i plug  r200 into computer, it will be all ok . 

I  only plug r200 into joule usb   , it can work ,but also slow . 

It can't work again when  i use usb-hub and plug in laser .

ROBOTIS-Will, 2017-07-31:
Hello ncnynl,

Depending on the system configuration and development environment, an external power supplied USB hub might be required in order to operate all components.
Please try with a power supplied USB hub.
Thank you.

ncnynl, 2017-08-07:
Hello  ROBOTIS-Will  

Yes  I have tried  other USB-HUB with power , but still not work.  
i will try other usb-hub again ! 

For this:
I only plug r200 into joule usb , it can work ,but also slow .

I solved this  by installing r200 driver from source instead deb package .  it work fine.

ROBOTIS-Will, 2017-08-13:
Hello ncnynl,

I'm sorry for the late reply due to my vacation last week.
Are you saying that reinstalling R200 driver resolved all problems you had?
Please let me know where you downloaded the driver so our software engineers can reproduce your case.
Thank you.

ncnynl, 2017-08-16:
Hi, ROBOTIS-Will  

Not all problems!  in the case of   i  inserted usb of R200 to intel joule  nor USB-HUB

i  install driver from debs through your tutorial  Before . like this :
sudo apt-get install ros-kinetic-librealsense
sudo apt-get install ros-kinetic-realsense-camera

this way slowed down the intel joule system.

i reinstalled from source later , like this :
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

this way work fine. 

And , i  tested by using jeston TX2 through USB-HUB + power cable, But it also was unable to work smoothly.

Thank you.

ROBOTIS-Will, 2017-08-21:
Hello ncnynl,

Thank you for sharing your solution for the issue.
SW engineer is currently examining the R200 issue and I will notify you as soon as he figures it out.
Thanks for your patience.

Han-IChun, 2017-10-13:
Hi all,
is there any progress on this issue?
I have a similar problem here.  Can't launch lidar and R200 simultaneously in Joule with USB hub but R200 works fine when it plugs into Joule directly. When using USB hub, terminal shows "No camera detected!"  

I've tried powered USB hub, but it didn't work for me.

Thanks,
I-chun

routiful, 2017-10-13:
Hi :)

We are checking to find what is make this issue.
We will be announce how to solve it soon.

Sorry for inconvenience. 

Thanks
Darby

ROBOTIS-Will, 2017-10-25:
Hello @ncnynl , @Han-IChun 
Could both of you check which BIOS version of Intel Joule you are using? BIOS 1J2 version seems unstable with Ubuntu.

Han-IChun, 2017-10-26:
Hi,
I use 1J2 version. 
Does 193 version launch R200 properly with usbhub? However, I had some problem on 193 version that I can't change the setting on BIOS. I don't sure if rollbacking to 193 version is a good idea for me.
Thank you for supports.
Best,
I-chun

robotpilot, 2018-02-12:
Has this problem been resolved?

robotpilot, 2018-02-13:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-13 by robotpilot