ROBOTIS-GIT/turtlebot3 #68: Really big bag file

allenh1, 2017-07-25:
In the turtlbot3_slam package, there's a really large bag file (68 MB) that makes it into the debian packages. It would be really great if that could be avoided, and replaced with a `catkin_download` instead.

routiful, 2017-08-21:
Hi :)

We changed STL files are more smaller than before.
Thank you to your advice.

Thanks
Darby

allenh1, 2017-08-21:
Could you remove the bag file too please? The `turtlebot_slam` package has a large bag file that gets included in the system binaries. You could change that to a `catkin_download_test_data` and host the file elsewhere.

robotpilot, 2018-03-13:
We removed large bag file in SLAM package. Thank you! :)
https://github.com/ROBOTIS-GIT/turtlebot3/releases/tag/0.2.0

closed 2018-03-13 by robotpilot

allenh1, 2018-03-13:
Thanks! :tada:

allenh1, 2018-03-13:
Thanks! :tada: