ROBOTIS-GIT/turtlebot3 #69: turtlebot3_core does not bringup correctly

Han-IChun, 2017-07-28:
In normal situation, after I do
`roslaunch turtlebot3_bringup turtlebot3_robot.launch`
it shows a lot of [INFO] after [INFO] connecting to /dev/ttyACM0 at 115200 baud, including subscriber and publisher build and sensor calibration.

But now, it stucks in [INFO] connecting to /dev/ttyACM0 at 115200 baud and noes not show anything else. (it is not always like this but very often) On remotePC, the ros_graph shows turtlebot3_core node is disconnected to others.

what should i do to solve it?

ROBOTIS-Will, 2017-07-31:
Hello Han-IChun,

The USB port or USB cable might have an issue in your case.
If you haven't modified the 115200 baudrate, I can think of below cases.

1. Make sure that your OpenCR - Raspberry Pi USB cable is securely connected. OpenCR USB port could be damaged due to an external force and this can cause irregular/unstable connection.
2. If your unplugged and plugged the USB cable again to Raspberry Pi 3 due to an error, ttyACM0 port could be occupied by previous process and the OpenCR could be assigned at ttyACM1 or some other port depending on your situation/configuration.

If you can take a full video of below procedure, it'd be very helpful for further trouble shooting.
1. Turn off the TurtleBot3.
2. Pull out the OpenCR-Raspberry Pi3 USB cable and plug in again.
3. Turn on the TurtleBot3.
4. Run programs.

Thank you.

Han-IChun, 2017-08-09:
Hello
thank you for the reply.
Yesterday I got a new Intel Joule board, so it should not be the USB port issue now. (supposed the new one is work correctly)
The first time bringup the turtlebot3, everything works fine in the beginning, but after about ten minutes test, the ros_graph on remotePC shows turtlebo3_core and trutlebot_lds nodes are disconnected to others.
I switch to SBC, terminal shows no error at that time. But when I press ctrl-C, some error occur. it is shown below.
[image]

After this error occur, I can not bringup turtlebot3 correctly anymore. I'v tried the procedure you mention above and there is no error, but it still not work. The bringup procedure stuck in  "[INFO] connecting to /dev/ttyACM0 at 115200 baud." 
[image]

Is there anyway to clear occupied port on both Intel Joule and OPEN-CR board?
I tried unplug and plug USB cables, but it's still not working. :(
thanks you

ROBOTIS-Will, 2017-08-21:
Hello Han-IChun,
The error "_[ERROR] [1503290964.906810386]: Error instantiating laser object. Are you sure you have the correct port and baud rate? Error was read: Interrupted system call_" could occur when terminating the basic bringup packages, but it is not a critical one.
When I normally run _roscore_ and _bringup_, what I see from rqt graph looks as below.
[image]
If you are still stuck at _Connecting to /dev/ttyACM0 at 115200 baud_ line, please type below command from your TurtleBot3 to see the port is available. Normally, you should only see _/dev/ttyACM0_ as a result.
**$ls /dev/ttyACM***
[image]
If you see multiple results, replace your USB cable and perform firmware recovery for OpenCR.
Please let us know if the issue remains.
Thank you.

Han-IChun, 2017-08-22:
Hi 
After updated BIOS 1J2, the USB port performed even weirder. The USB port did not work at all, it was locked that both keyboard and mouse fail. Due to a tight development schedule, I buy a new Intel Joule instead and sent the original one back to tb3 agent in Taiwan. They will do more tests and might feedback to you later. 
BTW, The new one works good. :")

Thank you for the supports.

I-CHUN

closed 2017-08-22 by Han-IChun

ROBOTIS-Will, 2017-08-22:
Glad to hear that the new one works well.
Thank you for letting us know about the status :)

moghulvasa, 2024-05-15:
roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch. I reached till this command for control turtlebot 3 burger using keyboard. After this , if i enter letter W,A,S,D,X Robot is not responding to my command. I checked the communication between OPEN CR and Raspberry pi, It's fine. Can anyone help me to sort out this problem
[image]