ROBOTIS-GIT/turtlebot3 #77: Intel joule 570X NO HDMI output

sisihi, 2017-08-24:
Hi robotis,

I encountered a problem when installing the ubuntu onto the intel joule
570x which is the core of turtlebot3 waffle:

`The HDMI can not output video signal to our monitors(samsung
LS24E390HK/XK).`

The monitor cant detect the board as well.

Changing three different HDMI cables, testing the board with
other monitors do not help.

Besides, the serial USB port output normally after upgrade the BIOS.

So, I'd like to know the specific monitor models that u completed the installation of unbuntu?

Thanks.

routiful, 2017-08-25:
Hi :)

I tested Joule board with latest Bios. ([Download Link](https://downloadcenter.intel.com/download/26948/BIOS-for-Intel-Joule-Compute-Module))

I give some tips for you.
First, you should power on the board after connect Type-C USB. Then you can check that power led will blink once.
Second, After flash the bios, you should disconnect Type-C USB and power off the board.
Lastly, you need to wait until connected monitor with the board. 

If this tips are not working on your problems, you would try old version of bios.

Please let us know if the problems remains.

Thanks
Darby

closed 2017-10-11 by routiful

rishabhagarwal880, 2017-11-06:
I am facing the same issue. I updated the bios 10 version 193 as recommended and once the bios was updated I was not able to recieve any hdmi output. I tried bios 1H3 too but the problem still exists. 

Please let me know if I am doing something wrong or the bios need to be updated to the latest version.

skercher, 2017-11-29:
I also am seeing this same issue.  I tried both firmware versions 1H3 and 1J2, but do not see any HDMI output....

LucasWEIchen, 2020-03-03:
same here,  got 2 joule. 1 HDMI stop working. tried all BIOS, all of them successfully flashed in to joule. STILL NO HDMI OUTPUT.