ROBOTIS-GIT/turtlebot3 #78: Costmap2D timeout problem

alirezaahmadi, 2017-08-29:
Hi 
I have implemented all instructions right as you've said in tutorials, and every thing was ok until I ran the navigation's example, which it loads the map but does not view the costmap2D and gives an warning like this:
[ WARN] [1503985578.700748532]: Costmap2DROS transform timeout. Current time: 1503985578.7007, global_pose stamp: 1503985533.6728, tolerance: 0.2000
I've changed the tolerance timeout but it did nothing on this issue. 
as I have this problem the robot does not move whenever I set a 2D Goal. (thats logical because the costmap2D does not exist !!!)
I would really appreciate if  anyone could help me to solve this problem.
Thanks

routiful, 2017-08-29:
Hi :)

Would you check frequency of tf topics using below commands.
$ rostopic hz /tf

OpenCR publishes tf topics in 30Hz. So, If the frequency is significantly different, check your OpenCR and network environments.

Thanks.
Darby

alirezaahmadi, 2017-08-29:
I checked it, responses like this:
average rate: 231.417
	min: 0.000s max: 0.117s std dev: 0.00784s window: 223

alirezaahmadi, 2017-08-29:
Thanks for answering ....

I checked it, it responses this way : 
alireza@ubuntu:~$ rostopic hz /tf
subscribed to [/tf]
average rate: 28.052
	min: 0.000s max: 0.146s std dev: 0.05273s window: 19
average rate: 140.834
	min: 0.000s max: 0.146s std dev: 0.01860s window: 243
average rate: 177.587
	min: 0.000s max: 0.146s std dev: 0.01330s window: 484
average rate: 194.925
	min: 0.000s max: 0.146s std dev: 0.01093s window: 726
average rate: 203.740
	min: 0.000s max: 0.146s std dev: 0.00954s window: 962
average rate: 211.368
	min: 0.000s max: 0.146s std dev: 0.00856s window: 1211
average rate: 216.040
	min: 0.000s max: 0.163s std dev: 0.00967s window: 1454
average rate: 218.366
	min: 0.000s max: 0.163s std dev: 0.01153s window: 1681
average rate: 220.356
	min: 0.000s max: 0.163s std dev: 0.01288s window: 1923

routiful, 2017-08-29:
In my enviroments, represents below:

average rate: 143.226
	min: 0.000s max: 0.192s std dev: 0.00634s window: 23515
average rate: 143.263
	min: 0.000s max: 0.192s std dev: 0.00636s window: 23665
average rate: 143.267
	min: 0.000s max: 0.192s std dev: 0.00637s window: 23812
average rate: 143.281
	min: 0.000s max: 0.192s std dev: 0.00636s window: 23960
average rate: 143.248
	min: 0.000s max: 0.192s std dev: 0.00635s window: 24100
average rate: 143.231
	min: 0.000s max: 0.192s std dev: 0.00635s window: 24242
average rate: 143.193
	min: 0.000s max: 0.192s std dev: 0.00633s window: 24380
average rate: 143.153
	min: 0.000s max: 0.192s std dev: 0.00632s window: 24516
average rate: 143.161
	min: 0.000s max: 0.192s std dev: 0.00631s window: 24664
average rate: 143.166
	min: 0.000s max: 0.192s std dev: 0.00630s window: 24811
average rate: 143.174
	min: 0.000s max: 0.192s std dev: 0.00630s window: 24839

Your average rate(Hz) of topic is higher than me. 
I'll give you some suggestions.

First, reset your OpenCR and restart RP3.
Second, change 'transform_tolerance' in costmap_common_params_burger.yaml 
Third, find better network condition

Thanks for letting me know how it goes.

Thanks
Darby

routiful, 2017-08-29:
I think average rate is fine.
How about reset your OpenCR and restart RP3?

alirezaahmadi, 2017-08-29:
I did some times but nothing changed !!!

routiful, 2017-08-29:
:)

Did you set time sync between your laptop PC and RP3?
This commands help you.

$ sudo apt-get install -y chrony ntpdate 
$ sudo ntpdate ntp.ubuntu.com

alirezaahmadi, 2017-08-29:
:) same result !!! 
is there any old package, which I could test??

alirezaahmadi, 2017-08-29:
I tested it without the RP3, and worked correctly!!
I think my problem is with my Network connection ... 
is there any important thing about Network?? 

Thanks ..

routiful, 2017-08-30:
:)

How to test it without the RP3? Did you use other SBC?
I think your network environments between PC and RP3 is fine. 
However, I doubt if there is a problem with the time sink.
Below link will helps you
[Time Sink](https://answers.ros.org/question/11180/chrony-configuration-and-limitations/)

Thanks
Darby

alirezaahmadi, 2017-08-30:
I mean, I tested the packages and nodes all running on one laptop... which they worked correcly.. I just wanted to make sure, there is no problem with them.. 
:)

routiful, 2017-09-01:
Okay, i see. Thanks.

Did you connect Wifi when you using RP3?

alirezaahmadi, 2017-09-01:
Yes I connect with Wifi.
I tried chrony, it did nothing, also I tested the time differences between two machines (PR3 and my laptop) it answered like this:
alireza@ubuntu:~$ ntpdate -q 192.168.1.3
server 192.168.1.3, stratum 0, offset 0.000000, delay 0.00000
31 Aug 21:04:18 ntpdate[31891]: no server suitable for synchronization found

I have to set the server to solve this warning , yes?

alirezaahmadi, 2017-09-02:
I have set servers on my laptop but, in RP3 the chrony gives my this error "506 Cannot talk to daemon" whenever I call its sources by "chrony sources". do you have any idea for this problem??

ltelang, 2017-11-02:
Ali,

Did you ever end up solving the error...
[ WARN] [1509638125.580617453]: Costmap2DROS transform timeout. Current time: 1509638125.5806, global_pose stamp: 1455209088.3206, tolerance: 1.0000

I have the exact same problem as you...went through the right steps for bringup, teleoperation etc.. but then I come to set a 2D nav goal and it gives me that error.  My 2D coast map has to be missing.  I am unsure how to manipulate the frequency that you talked about above

please help!

alirezaahmadi, 2017-11-02:
actually, no... I just tried to make all system clocks same... and it worked, but it wasn't a stable solution.. chrony doesn't work on raspberry pi, I don't know why... and it's gave me the error "506 Cannot talk to daemon"..

ltelang, 2017-11-02:
So I just went in manually on my RP3 went System menu>Administration>Time and Date> unlocked it using my password and changed the time. 

It got rid of that Cost2DROS transform timeout error and I get a new one:
[ WARN] [1509644376.618270534]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101057 timeout was 0.1.

This might not even be progress, i have no clue, but it's something different have you tried just manually changing it?

jrjennings, 2017-11-03:
I have ran into the same problem with my turtlebot 3 ,  the time difference is huge and it always happens when the network doesn't have an internet connection but work fine when I do.  

after running
roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map1.yaml

any ideas?

jrjennings, 2017-11-08:
Adding a real time clock worked for me.  https://https://www.mintsim.com/-mate.community/t/supporting-ds3231-rtc-source-synchronise-with-network-time/5711

ltelang, 2017-11-09:
I am going to try adding the real clock and then get back to you on this jrjennings

ltelang, 2017-11-09:
I have ordered a couple of the DS3231 chips, so it may take a couple days before I can test this

cs2r, 2017-11-14:
I have the same problem with turtlebot33-waffle any solutions???

robotpilot, 2018-02-12:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-12 by robotpilot

tinsky, 2018-05-23:
I have the same problem with turtlebot3-waffle. any solutions?

routiful, 2018-05-23:
Hi :)

Check this FAQ in WIKI

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remotepc

Thanks.

tinsky, 2018-05-24:
thx for your reply!
I have tried ntpdate to synchronize time of turtlebot and remote PC. but it did nothing.
To make sure the time of turtlebot and remote PC is synchronized, I even set up a ntp server on the remote PC and use ntpd on turtlebot to update time. but it also did nothing.

Maybe is the problem with my network connection? 
Because I don't have any good enough Wi-Fi to connect, I set up an Access Point mode Wi-Fi hotspot on my remote PC and make turtlebot connect the hotspot.
So every time I run the navigation's example, the network doesn't have the Internet connection.

any ideas?

appreciate your help!

jrjennings, 2018-05-24:
Have you tried using chrony to synchronize?  I've been running my burger without an internet connection using chrony.  I had to do some fiddling with the config files to get it working. Isolated networks at the bottom of this page was helpful https://chrony.tuxfamily.org/doc/3.3/chrony.conf.html

tinsky, 2018-05-25:
thx for your advice!
I will try it as soon as possible to see whether it works for my waffle.
appreciate your help!

tinsky, 2018-05-28:
@jrjennings 
it does works!! thank you very much.

ProfJust, 2021-01-15:
I had the same issue using the **SR04-Sonar-Sensor** . 
Therefore I uncommented the sonar-functions in the turtlebot-core-source in line 145 etc.

With 
`$ rostopic hz /tf`
you can check the frequency. 
If the Distance is to long or there is no Sensor connected the /tf-frequency decreases to **below 1Hz.**

LOKAKSHAI, 2023-07-19:
I'm also getting this error

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

 

WARNING The following node subscriptions are unconnected:
* /amcl:
   * /initialpose
* /move_base:
   * /move_base_simple/goal
   * /map_updates

 

Found 1 error(s).

 

ERROR The following nodes should be connected but aren't:
* /move_base->/move_base (/move_base/global_costmap/footprint)
* /move_base->/move_base (/move_base/local_costmap/footprint

I tried changing the global frame from map to odom, but the error still remains. How to fix this issue