ROBOTIS-GIT/turtlebot3 #79: Laser_scan and Robot_model error in RViz SLAM

idrobotiskidslab, 2017-09-12:
Hello,

My next question is about
My turtlebot 3 burger using Ubuntu Mate 16.04
My remote PC using Ubuntu Xenial 16.04
My ROS using Kinetic Kame
Can you send me some links or file with extension .img that contain ubuntu xenial 16.04 and ROS kinetic kame for my turtlebot3 ?

Because if i do the same like tutorial from turtlebot3 official website (http://turtlebot3.readthedocs.io/en/latest/slam.html) my turtlebot 3 burger cannot use SLAM and NAVIGATION feature from it with error message below. I think there is some update from ROS Kinetic Kame or  Ubuntu xenial 16.04 or Ubuntu Mate 16.04 one of them that caused the problem. Can you give me some default package for my turtlebot 3 Burger?

Or  can you  guide me directly to resolve this problem?
Thank you...
[image]

robotpilot, 2017-09-12:
Unfortunately, we can't provide Ubuntu img file. Commercial redistribution
of Ubuntu is a license violation.
Instead, explain the problem in detail, and you can support you.

When you run a file, is there a problem like the picture file you sent?

-- 
//------------------------------------------------------------------------------
Yoonseok Pyo, Ph.D.,
Research Engineer / Open Source Team / R&D Dept.
ROBOTIS CO., LTD. (http://www.robotis.com)
#1505 Ace High-End Tower NO. 3, Gasan-dong 145,
GasanDigital 1-ro, GeumCheon-gu, Seoul, Korea 153-787

Follow ROBOTIS on Twitter <https://twitter.com/robotis>, Facebook
<https://www.facebook.com/robotis.company>, Linkedin
<https://www.linkedin.com/company/robotis>, Github
<https://github.com/ROBOTIS-GIT>
Personal Website (http://www.robotpilot.net)
//------------------------------------------------------------------------------

idrobotiskidslab, 2017-09-13:
thank you for your helping, my Turtlebot3 Burger can SLAM and Navigation now. But I confuse the problem cause. I just type 'apt-get update' and 'apt-get upgrade' and problem has solved. Some days ago I do that but still can't SLAM and Navigation, maybe about Network Environment.

robotpilot, 2017-09-13:
Many people have trouble with this because of the time difference between
Remote PC and TurtleBot3 PC.
This is done by matching the time servers to each other, as shown in the
following command.

$ sudo ntpdate ntp.ubuntu.com

This setting can be done automatically if the Internet is properly
connected when the computer is running without having to do it every time.
Like your case, I think.

Thanks,
Pyo

-- 
//------------------------------------------------------------------------------
Yoonseok Pyo, Ph.D.,
Research Engineer / Open Source Team / R&D Dept.
ROBOTIS CO., LTD. (http://www.robotis.com)
#1505 Ace High-End Tower NO. 3, Gasan-dong 145,
GasanDigital 1-ro, GeumCheon-gu, Seoul, Korea 153-787

Follow ROBOTIS on Twitter <https://twitter.com/robotis>, Facebook
<https://www.facebook.com/robotis.company>, Linkedin
<https://www.linkedin.com/company/robotis>, Github
<https://github.com/ROBOTIS-GIT>
Personal Website (http://www.robotpilot.net)
//------------------------------------------------------------------------------

KaiqiaoTian, 2017-11-03:
Hi I'm very confuse how turtlebot3 connect with remote PC? There is no wifi Modula in my turtlebot3. I can connect Buletooth  with remote pc, is it work?

I want to learn turtlebot3 with a group. Do you guys have a study group? 
Thanks,
kaiqiao

robotpilot, 2017-11-06:
@idrobotiskidslab 
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.

@KaiqiaoTian 
Your answers are being handled in other issues. #88

closed 2017-11-06 by robotpilot