ROBOTIS-GIT/turtlebot3 #80: Successfully roslaunch in lidar and core, but cannot see point clouds in RVIZ   (No error message)

ShirleyZYJ, 2017-09-28:


ROBOTIS-Will, 2017-09-29:
@ShirleyZYJ 
Hello,
Could you list up the commands you entered to run the Turtlebot3?
Were you able to see the 3D graphical image of Turtlebot3 on the RVIZ screen like this?
[image]

ShirleyZYJ, 2017-09-29:
@ROBOTIS-Will 
Thanks for your reply.
Yesterday's problem might due to the port interruption between each other, I have solved that by using a USB hub to connect the keyboard, mouse and USB from arduino. Currently, it can function well. But it is strange indeed, when I made the connection just like the tutorial said , it doesn't work.

jediofgever, 2017-10-25:
Hey Sorry for opening this issue again, I am having exactly the same issue. I did all steps as official guide however I am not able to see a map.  Below picture is screenshot of my Rviz envoirment could you please look at it ? 
[image]

routiful, 2017-10-25:
Hi :)

Do you check messages are communicated between Remote PC and TB3?
I give some link to check it.

[Remote PC](http://turtlebot3.robotis.com/en/latest/pc_software.html#network-configuration) and [TB3](http://turtlebot3.robotis.com/en/latest/sbc_software.html#network-configuration-burger-and-waffle)

Thanks
Darby

jediofgever, 2017-10-25:
Hi @routiful , 
I have double checked that both steps and they can communicate well. I can successfully do keyboard teleoperation from remote PC. So that means I have set up the IP adresses correctly however when I try to do SLAM manipulation, the Rviz shows up and nothing happens. the screenshot I provided, it says `No map recieved`. I have done time sync on both Remote PC and TB3. and lastly when I visulize the map on terminal there is some warning which may  help to solve the problem. here is that warning ; 
[image]

ShirleyZYJ, 2017-10-25:
Hi @jediofgever 

I guess we met the same problem, can you see the red point cloud info at the bring up stage?

Shirley

jediofgever, 2017-10-25:
Hi @ShirleyZYJ ,
Unfourtunately I am not seeing point clouds as well here is what I got from bringup stage ;
[image]

ShirleyZYJ, 2017-10-25:
@jediofgever 
Wow, we are exactly on the same stage then. 
My system is also not very stable currently, but you may try the following step:
1. Restart your PC and TB3
2. Before roscore use      sudo ntpdate ntp.ubuntu.com      to synchronize the time on both PC and TB3
Then you can follow the tut to bring up your robot.

This tip might help, but I cannot guarantee, for it is not always work on my robot either. I am still trying to figure out the final solution

ShirleyZYJ, 2017-10-25:
@jediofgever 

BTW please let me know whether this tip work or not thanks.

Shirley

jediofgever, 2017-10-25:
@ShirleyZYJ  
I did exactly what you said however still didnt get it work. One thing I noticed the  `sudo ntpdate ntp.ubuntu.com` command gives different outputs on Remote PC and TB3 Can you show me the output of your Remote PC and TB3. Are they the same ? 
On TB3 i got output as ; 
``
On Remote PC I got output as ;
`25 Oct 17:08:44 ntpdate[3336]: adjust time server 91.189.91.157 offset -0.000558 sec`

ShirleyZYJ, 2017-10-25:
@jediofgever 

Do you mean that you do not have any output on TB3?
My output on TB3 and PC are almost in the same format. The offset for PC is just like yours or at a 0.001 scale but the offset for TB3 is much more larger, usually on a 0.1 scale.

Shirley

jediofgever, 2017-10-25:
@ShirleyZYJ '
Actually I have output from TB3 too, sorry I just fixed it. I edited my comment, do you think the `sudo ntpdate ntp.ubuntu.com` gives proper outputs in my case ?

ShirleyZYJ, 2017-10-25:
@jediofgever 

Yes I guess the output is fine, but your problem is not solved.
I have open another issue to look into this problem here https://github.com/ROBOTIS-GIT/turtlebot3/issues/83 , but unfortunately it is also not solved.
There are some other tips in this issue you may interest and I will update the issue once I found the solution.

BTW I have one more question, did you ever make it work one or two times or it never goes fine since you get the robot.

Shirley

jediofgever, 2017-10-25:
I have reviewed whole issues but I have not make it work even once unfortunately. I am stuck in step ten which is SLAM. That would be great if you share solution once you find it.

ROBOTIS-Will, 2017-10-25:
@jediofgever 
ntpdate command helps to match time between PC and TB3.  
When two system have different time setting, time sensitive data could be recognized as "not fresh" and be ignored which can cause similar issue you are experiencing now.  
If time sync does not help, you might have corrupted 3D modeling file for Turtlebot.  
Have you tried reinstalling packages?

jediofgever, 2017-10-25:
Hi @ROBOTIS-Will,
I tried ntpdate but it doesnt seem to help me. I am not sure if I have corrupted file of Turtlebot. Could you be more specific about that packages ? 
I have tried in Gazebo and it works fine ? I dont know what does this mean. What can be reason Do I have a hardware problem then ?

ROBOTIS-Will, 2017-10-25:
@jediofgever 
Sorry about the unclear explanation.  
I was talking about re-downloading the 3D modeling file saved in the "turtlebot3_description\meshes" folder.
Or, you might want to try this method(https://gist.github.com/awesomebytes/2595b1dc41831c804a4f)

jediofgever, 2017-10-26:
@ROBOTIS-Will ,
I reinstalled that package and not working. I also tried the method (https://gist.github.com/awesomebytes/2595b1dc41831c804a4f), the warnings I got before now disappeared but still no map recieved says Rviz. 
I run this command `rosrun rviz rviz -d `rospack find turtlebot3_slam`/rviz/turtlebot3_slam.rviz
`
and the output only gives these lines and stucks 
`[ INFO] [1509012574.314191941]: rviz version 1.12.13
[ INFO] [1509012574.314313059]: compiled against Qt version 5.5.1
[ INFO] [1509012574.314346025]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1509012574.781056623]: Stereo is NOT SUPPORTED
[ INFO] [1509012574.781143392]: OpenGl version: 3 (GLSL 1.3).
`

ROBOTIS-Will, 2017-10-30:
@jediofgever 
Just to make sure, did you successfully run the `su ntpdate ntp.ubuntu.com` on TB3? 
I missed your edited posting that says ntp socket was in use and exited without running it on TB3.
You have to run ntpdate command on both Remote PC and TB3.
You might have different numbers in the output.

jediofgever, 2017-10-30:
@ROBOTIS-Will , 
` su ntpdate ntp.ubuntu.com` gives ;
`the NTP socket is in use, exiting` but I sucessfully run `ntpdate -u pool.ntp.org` command on both PC and TB3. It did not work problem is still there

robotpilot, 2018-02-12:
@jediofgever 
I'm sorry we don't have any action. Has this problem been resolved?

robotpilot, 2018-02-13:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-13 by robotpilot

YXGuan, 2022-03-13:
Hi there, @robotpilot @ROBOTIS-Will, I am having the same issue of no point cloud in RVIZ and would really appreciate your help. I have already tried all the method listed in the previous discussion, but unfortunately they didn't resolve the issue.

[image]

[image]

[image]

Below are the commands:

On Rpi (turtlebot)
`ubuntu@ubuntu:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/ubuntu/.ros/log/0d2ab0ae-a2f6-11ec-8a33-e5b4ca0c84b0/roslaunch-ubuntu-2201.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.35:44559/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.113:11311

process[turtlebot3_core-1]: started with pid [2215]
process[turtlebot3_lds-2]: started with pid [2216]
process[turtlebot3_diagnostics-3]: started with pid [2217]
[INFO] [1647193882.252366]: ROS Serial Python Node
[INFO] [1647193882.304433]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1647193884.426053]: Requesting topics...
[INFO] [1647193884.458783]: Note: publish buffer size is 1024 bytes
[INFO] [1647193884.471666]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1647193884.492314]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1647193884.569279]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1647193884.584459]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1647193884.625106]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1647193884.644411]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1647193884.664312]: Setup publisher on battery_state [sensor_msgs/BatteryState]
[INFO] [1647193884.685419]: Setup publisher on magnetic_field [sensor_msgs/MagneticField]
[INFO] [1647193884.758115]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1647193884.790505]: Note: subscribe buffer size is 1024 bytes
[INFO] [1647193884.801146]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1647193884.833735]: Setup subscriber on sound [turtlebot3_msgs/Sound]
[INFO] [1647193884.862379]: Setup subscriber on motor_power [std_msgs/Bool]
[INFO] [1647193884.893719]: Setup subscriber on reset [std_msgs/Empty]
[INFO] [1647193887.552330]: Setup TF on Odometry [odom]
[INFO] [1647193887.563159]: Setup TF on IMU [imu_link]
[INFO] [1647193887.573526]: Setup TF on MagneticField [mag_link]
[INFO] [1647193887.584157]: Setup TF on JointState [base_link]
[INFO] [1647193887.599044]: --------------------------
[INFO] [1647193887.609962]: Connected to OpenCR board!
[INFO] [1647193887.620240]: This core(v1.2.6) is compatible with TB3 Burger
[INFO] [1647193887.630881]: --------------------------
[INFO] [1647193887.641455]: Start Calibration of Gyro
[INFO] [1647193890.110545]: Calibration End
`

on laptop:
`roscore
yuxiang@yuxiang:~$ rosrun rviz rviz -d `rospack find turtlebot3_description`/rviz/model.rviz
[ INFO] [1647194518.736690919]: rviz version 1.14.14
[ INFO] [1647194518.736766558]: compiled against Qt version 5.12.8
[ INFO] [1647194518.736790377]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1647194518.778046429]: Forcing OpenGl version 0.
[ INFO] [1647194518.967926089]: Stereo is NOT SUPPORTED
[ INFO] [1647194518.967988556]: OpenGL device: Mesa Intel(R) UHD Graphics (CML GT2)
[ INFO] [1647194518.968008862]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
`