ROBOTIS-GIT/turtlebot3 #83: No map error in RVIZ

ShirleyZYJ, 2017-10-10:
The thing is I have successfully roslaunching on lidar, but I cannot see the point clouds in RVIZ, their is no error message found in terminal, and the RVIZ shows "no map"(picture 1) and there is no position info in the slam terminal (picture 2 column 2).

Yesterday it functioned well actually (picture 3) and I don't know what goes wrong today. Actually, it is not my first time to met this problem, last time it is solved by restart two computer (Remote PC and turtlebot), but this time it doesn't work either. Is there any solution to this problem?
[image]
[image]
[image]

routiful, 2017-10-11:
Hi :)

Can you show me an error message when you launch a turtlebot3_robot.launch file?
It makes more help me for your issue.

Thanks
Darby

ShirleyZYJ, 2017-10-11:
**@routiful 
And if you mean the result after slam launch command =>roslaunch turtlebot3_slam turtlebot3_slam.launch
The result is here:**

... logging to /home/daodao/.ros/log/acf27066-ae42-11e7-8658-681729f2b327/roslaunch-daodao-Inspiron-5721-11216.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://192.168.1.149:35549/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /turtlebot3_slam_gmapping/angularUpdate: 0.2
 * /turtlebot3_slam_gmapping/astep: 0.05
 * /turtlebot3_slam_gmapping/base_frame: base_footprint
 * /turtlebot3_slam_gmapping/delta: 0.05
 * /turtlebot3_slam_gmapping/iterations: 5
 * /turtlebot3_slam_gmapping/kernelSize: 1
 * /turtlebot3_slam_gmapping/lasamplerange: 0.005
 * /turtlebot3_slam_gmapping/lasamplestep: 0.005
 * /turtlebot3_slam_gmapping/linearUpdate: 0.2
 * /turtlebot3_slam_gmapping/llsamplerange: 0.01
 * /turtlebot3_slam_gmapping/llsamplestep: 0.01
 * /turtlebot3_slam_gmapping/lsigma: 0.075
 * /turtlebot3_slam_gmapping/lskip: 0
 * /turtlebot3_slam_gmapping/lstep: 0.05
 * /turtlebot3_slam_gmapping/map_update_interval: 2.0
 * /turtlebot3_slam_gmapping/maxUrange: 4.0
 * /turtlebot3_slam_gmapping/minimumScore: 100
 * /turtlebot3_slam_gmapping/odom_frame: odom
 * /turtlebot3_slam_gmapping/ogain: 3.0
 * /turtlebot3_slam_gmapping/particles: 120
 * /turtlebot3_slam_gmapping/resampleThreshold: 0.5
 * /turtlebot3_slam_gmapping/sigma: 0.05
 * /turtlebot3_slam_gmapping/srr: 0.01
 * /turtlebot3_slam_gmapping/srt: 0.02
 * /turtlebot3_slam_gmapping/str: 0.01
 * /turtlebot3_slam_gmapping/stt: 0.02
 * /turtlebot3_slam_gmapping/temporalUpdate: 0.5
 * /turtlebot3_slam_gmapping/xmax: 10.0
 * /turtlebot3_slam_gmapping/xmin: -10.0
 * /turtlebot3_slam_gmapping/ymax: 10.0
 * /turtlebot3_slam_gmapping/ymin: -10.0

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    turtlebot3_slam_gmapping (gmapping/slam_gmapping)

ROS_MASTER_URI=http://192.168.1.149:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [11237]
process[turtlebot3_slam_gmapping-2]: started with pid [11238]

**The warning message should not influence the final result I guess. But there should be some position info after "started with pid" right?**

routiful, 2017-10-11:
Hi :)

First, check lidar is working fine
bringup lidar and open rviz by using under commands

[RP3]
`$ roslaunch turtlebot3_bringup turtlebot3_lidar.launch`

[REMOTE PC]
`$ roslaunch turtlebot3_bringup turtlebot3_model.launch `

Second, check slam is working properly
Do you check below text after _started with pid_ ?
```
[ INFO] [1507705371.862143669]: Laser is mounted upwards.
 -maxUrange 4 -maxUrange 3.49 -sigma     0.05 -kernelSize 1 -lstep 0.05 -lobsGain 3 -astep 0.05
 -srr 0.01 -srt 0.02 -str 0.01 -stt 0.02
 -linearUpdate 0.2 -angularUpdate 0.2 -resampleThreshold 0.5
 -xmin -10 -xmax 10 -ymin -10 -ymax 10 -delta 0.05 -particles 120
[ INFO] [1507705371.867288038]: Initialization complete
update frame 0
update ld=0 ad=0
Laser Pose= -0.0315088 -0.00424469 -3.00562
m_count 0
Registering First Scan
update frame 3
update ld=1.27415e-05 ad=7.77054e-05
Laser Pose= -0.031509 -0.00424299 -3.00567
m_count 1
Average Scan Matching Score=259.252
neff= 120
Registering Scans:Done
update frame 6
update ld=5.69363e-06 ad=0.000177977
Laser Pose= -0.0315088 -0.00424497 -3.00561
m_count 2
^C[turtlebot3_slam_gmapping-2] killing on exit
[robot_state_publisher-1] killing on exit
Average Scan Matching Score=263.97
neff= 120
Registering Scans:Done
```
If not, please open rqt and check tf tree.

Thanks
Darby

ShirleyZYJ, 2017-10-11:
@routiful 
Thanks~
I have followed your step to check.
The lidar is fine. but there is no text like above after "_started with pid_", I guess it is reason.
So how can I solve this problem

routiful, 2017-10-11:
Then, Can you try SLAM in [Gazebo](http://turtlebot3.robotis.com/en/latest/simulation.html#gazebo-3d)?? 
I want to know exactly what is wrong about SLAM package.

Thanks
Darby

ShirleyZYJ, 2017-10-11:
**The result of roslaunch in gazebo is following:**

... logging to /home/daodao/.ros/log/96dbb994-ae61-11e7-8658-681729f2b327/roslaunch-daodao-Inspiron-5721-17303.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://192.168.1.149:36361/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_urdf (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://192.168.1.149:11311

core service [/rosout] found
process[gazebo-1]: started with pid [17324]
process[gazebo_gui-2]: started with pid [17329]
process[spawn_urdf-3]: started with pid [17334]
[ INFO] [1507712454.718602106]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1507712454.719098288]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1507712455.314102407, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1507712455.409733450, 0.116000000]: Physics dynamic reconfigure ready.
[ INFO] [1507712456.833463885, 0.940000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1507712456.833609055, 0.940000000]: Starting Laser Plugin (ns = /)
[ INFO] [1507712456.852088628, 0.940000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1507712456.964305471, 0.940000000]: Starting plugin DiffDrive(ns = //)
[ WARN] [1507712456.964488698, 0.940000000]: DiffDrive(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1507712456.966232712, 0.940000000]: DiffDrive(ns = //): <tf_prefix> = 
[ WARN] [1507712456.966992256, 0.940000000]: DiffDrive(ns = //): missing <odometrySource> default is 1
[ WARN] [1507712456.967115239, 0.940000000]: GazeboRosDiffDrive Plugin (ns = ) missing <publishTf>, defaults to 1
[ INFO] [1507712456.968851006, 0.940000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1507712456.970648621, 0.940000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1507712456.986692860, 0.940000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1507712456.987810249, 0.940000000]: DiffDrive(ns = //): Advertise odom on odom 
[ INFO] [1507712457.016700805, 0.940000000]: imu plugin missing <frameName>, defaults to <bodyName>
[spawn_urdf-3] process has finished cleanly
log file: /home/daodao/.ros/log/96dbb994-ae61-11e7-8658-681729f2b327/spawn_urdf-3*.log

and in the gazebo terminal, the following info shows repeatedly
[image]

Thanks
Shirley

routiful, 2017-10-12:
I think that data time sync is not correctly worked.
As shown in the following command to match time sync

`$ sudo ntpdate ntp.ubuntu.com`

Thanks
Darby

ShirleyZYJ, 2017-10-12:
@routiful 
I just tried the time sync command to match the time. But I still cannot see the point cloud and no position info, I guess time might not be the problem.

Thanks
Shirley

routiful, 2017-10-12:
This issue was handled in ROS ANSWERS.
I give some link to you for resolve problem.

http://wiki.ros.org/tf/Errors%20explained

https://answers.ros.org/question/201948/tf_old_data-ignoring-data-from-the-past-for-frame-odom/

https://answers.ros.org/question/206748/tf_old_data-ignoring-data-from-the-past-for-frame-openni_depth_optical_frame/

Thanks
Darby

ShirleyZYJ, 2017-10-12:
I have carefully go through all these links and tried to synchronize the time,but the tf tree still shows like this while running the lidar.
[frames.pdf](https://github.com/ROBOTIS-GIT/turtlebot3/files/1378568/frames.pdf)

By the way, when I use the command **rosparam set use_sim_time true** to set up the clock, the situation will be even worse, the rviz even cannot show the actual pos of TB3 itself.

Thanks 
Shirley

ShirleyZYJ, 2017-10-16:
This problem is truly about system time synchronize.
Remove this line from the launch file might help <param name ="/use_sim_time" value="true"/>

routiful, 2017-10-16:
Hi :)

I can't see it which line make this problem.
Can you show me how to solve it?

Thanks
Darby

ShirleyZYJ, 2017-10-16:
This line <param name="use_sim_time" value="true"/>
Remove this line works for me, but I also see some solved issued that adding this line to launch file or set value to true might help. 

Thanks
Shirley

routiful, 2017-10-25:
Then, how about your TB3? 
Still it can't find scan data??

ShirleyZYJ, 2017-10-25:
@routiful 
Sad, it can work sometimes, but not very stable. Remove this line "param name="use_sim_time" value="true"/" might not be the final solution.
I still cannot figure out what cause this unstable problem. Sometimes restart might help, but sometimes not. 
BTW, I have add that line back to the launch file and I will do time syc each time before starting roscore by "sudo ntpdate ntp.ubuntu.com".
Hope this unstable problem can be solved later and I really want to know what causes this.

routiful, 2017-12-21:
If you have any problem with this issue, please reopen it.

Thanks
Darby

closed 2017-12-21 by routiful