ROBOTIS-GIT/turtlebot3 #84: Questions on TB3 Burger

cyberobotics, 2017-10-11:
I have some questions below relating to TB3 burger as follows:

1. How much time for the burger to map an area of 1000 sq ft?
2. Any area size limit? Is it possible to map multi level building?
3. Can the burger be able to map on its own?
4. We watched youtube and knew that it is able to go from point A to point B, Can we preset it to go in sequence from A to B then to C... and then to D...etc ?   Also can we use a device in a specified area that can call the burger to come? 
5. We have no problem to add a Camera for obstacle avoidance, is it possible for the burger send photos and view area in real time when the user is traveling.
6. Can the burger be able to work on its own when the remote PC is off?

Thanks.

routiful, 2017-10-11:
Hi :) @cyberobotics 

1. Estimate time to make map using burger depends how complex environment for example chairs, tables, thresholds and etc. Considering motor speed and range of lidar, in my opinion, burger takes 20~30 min to make map an area of 1000 sq ft.

2. There is no size limit. we have tested whole level in our office building previously. 
https://youtu.be/lkW4-dG2BCY

I'm sorry. I don't understand your question when TB3 makes map in multi level building. But slam_gmapping package only supports 2d occupancy grid map. 

3. We are not support automated slam with TB3

4. Yes, you can. We already presented turtlebot3_deliver package at ICRA2017
May this [source codes](https://github.com/ROBOTIS-GIT/turtlebot3_deliver/tree/master/turtlebot3_carrier_icra2017/src) help you. 

5. Of course, you can use raspberry pi camera and [open source ](https://discourse.ros.org/t/raspberry-pi-camera-node/1388). Show an [example video ](https://youtu.be/VYlMywwYALU)

6. Unfortunately, we does not suggest that. If you want to do that, we highly recommend to use other single board computer like Intel Joule.

Thanks
Darby

cyberobotics, 2017-10-12:
Hi Darby,

Thanks for the reply.  We have some follow up questions as below:

For questions #2, it is great to map the area with the size of the floor (eg. 15 floor), so how many floors can the TB3 be mapped? That is what I mean for mapping in multi-level building. 

For # 6, which means that TB3 Waffle is able to work on its own, right?
I would like to elaborate the main functions of the product we want to develop based on TB3 platform.
We want the unit act as a Home security Robot, the unit can automatically patrol all the rooms in the house in a particular time that the user preset.
The TB3 is able to send message, videos or photos to the user's mobile phone if there is something happen in a specified area.
On the other hand, the user is able to override to control the unit  even the user is traveling.
Of course, the unit is capable to avoid obstacle during its patrol. Docking for recharge is also needed. 
Is the TB3 Waffle or Burger able to do that?

Thanks.

routiful, 2017-10-13:
Hi :)

Q2. I understand. TB3 can map unlimited space. But the map have to be generated each floor separately (The map is not linked)

Q6. Awesome!! In my opinion, Waffle is more proper than Burger. Waffle has nice camera (realsense r200), powerful computer (joule) and large payload compared by Burger. Moreover, Waffle can operate all function itself ;)
The docking service is being developed. It will be released Nov. (include Waffle and Burger)

Thanks
Darby

robotpilot, 2017-11-06:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-11-06 by robotpilot

DSaMC, 2018-01-26:
Hi !

Any updates about a docking service?

Thanks!

robotpilot, 2018-01-27:
Our team recently provided a solution for docking. 
We use the intensity and distance of laser distance sensor and a reflective tape in this solution. Check the link below. :)
http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#automatic-parking
https://youtu.be/IRtdxoPo8Y8