ROBOTIS-GIT/turtlebot3 #85: Intel Joule 570x slower than the turtle

jediofgever, 2017-10-21:
I just got the turtlebot3 and I am stuck with step 6, SBC software setup because of slowness of the board. It cant even show ubuntu desktop, the cpu usage is close to zero, there is not even a single program running yet the board will get freeze and not respond to keyboard , mouse. Please provide me some help if you can thank you very much

ROBOTIS-Will, 2017-10-25:
@jediofgever 
Hello, could you check your BIOS version for Joule? If you have updated to 1J2(Latest version), you should rollback to the 193 version.  
Ubuntu may not run properly on Joule with 1J2 firmware.

robotpilot, 2017-11-06:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-11-06 by robotpilot