ROBOTIS-GIT/turtlebot3 #86: I want to know how Turtlebot3(turtlebot3_teleop_key) & Cartographer ROS

RyuSangGi, 2017-10-23:
Hi
I saw turtlebot3 & Google Cartographer ROS on the site( http://turtlebot3.robotis.com/en/latest/slam.html)
I want to manipulate directly to make a SLAM map. 
For example, I want to link cartographer instead of Gmapping on this site.(http://turtlebot3.robotis.com/en/latest/slam.html)... turtlebot3_teleop_key + Cartographer

It is not good when applied.
The topic is not activated in turtlebot3_teleop_key node because Cartographer ROS is applied.
I refer to this site and install and apply it. ( https://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/ ) 
The demo of this site was successful.

Please, Can you teach me how to apply it?

robotpilot, 2017-10-23:
Hi @RyuSangGi 

We made a demonstration(https://youtu.be/lkW4-dG2BCY) of whether a Cartographer can be used with TurtleBot3. It was tested only with SLAM using Cartographer. The official SLAM algorithm of TurtleBot3 is still Gmmaping(http://wiki.ros.org/gmapping).

The Cartographer is not yet available for all TurtleBot series, although some of Cartographer tips can be found on the link you mentioned. Until end of this year, we have a plans to change from Gmmaping to Cartographer, but we can not offer it right away. I'm sorry we can't support you right away.

We can't provide a complete solution right away, but if you have any difficulties, please tell us, then we can provide technical support.

robotpilot, 2017-11-06:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.

samby707, 2018-02-19:
I have created a gazebo environment with turtle bot. I want to use cartographer slam on the fly with rviz. Meaning when i give a positon and orientation in r viz as the end goal to the turtle bot, the turtle bot should move,create the slam and do autonomous navigation without hitting anywhere using cartographer. It is possible with gmapping. But can you tell me how to do it using cartographer?

ncnynl, 2018-03-17:
I have tested using cartographer for burger， it is all ok now . later i will share the code

ncnynl, 2018-03-18:
@robotpilot  @RyuSangGi   the code for cartographer is here : https://github.com/ncnynl/turbot3

robotpilot, 2018-03-18:
Recently, many users have been asked to make official TurtleBot3 packages that support Cartographer. We are trying to resume this support work.

Thanks, :)

robotpilot, 2018-05-17:
Hi @RyuSangGi

I added a Cartographer with the help of @ncnynl. We are now updating to the develop branch. I will release it in the next TurtleBot3 software release. https://github.com/ROBOTIS-GIT/turtlebot3/tree/develop/turtlebot3_slam/launch

robotpilot, 2018-05-25:
Hi, again!

We update and release new firmware (version 1.2.0) and new software (version 1.0.0) for users of TurtleBot3. :) 

Thank you to your contribution!

Please refer to the link below for details of this update.

https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888

closed 2018-05-25 by robotpilot