ROBOTIS-GIT/turtlebot3 #87: Follower and Teleoperation

MuratCA77, 2017-10-24:
I was able to control TurtleBot 3 robot using my keyboard commands (teleoperation) first. Then, I decided to try follower application demo and followed the instructions about installing packages. Then, unfortunately, follower demo did not work. What is worse is that when I tried to go back to teleoperation, unfortunately, I could not control the robot this time using keyboard commands.

Could you please give me your guidance on this issue?

Thanks,
Mark

routiful, 2017-10-25:
Hi :)

May you have changed _turtlebot3_lidar.launch_ file for follower demo. [(LINK)](http://turtlebot3.robotis.com/en/latest/applications.html#turtlebot-follower-demo)
It makes trouble to use teleoperation. 

Can you explain more specific situation and development enviroment or error code on your terminal to me while TB3 is launched following demo?
It will be make me help you easliy

Thanks
Darby

MuratCA77, 2017-10-25:
Yes, I have changed turtlebot3_lidar.launch file from base_scan to odom for the follower demo. Then, I changed it back to base_scan but teleoperation did not work this time.

MuratCA77, 2017-10-25:
Actually, I have just fixed the issue. But somehow, follower demo is not functioning properly.

routiful, 2017-12-21:
If you have any problem to launch follower demo, please reopen issue.

Thanks
Darby

closed 2017-12-21 by routiful