ROBOTIS-GIT/turtlebot3 #88: How turtlebot3 connect with remote PC?

KaiqiaoTian, 2017-11-03:
Hi, I just got my turtlebot 3 , and confuse by the wifi connect with remote PC.

Pls help.

Thanks,
kaiqiao

plguhur, 2017-11-03:
Well this is pretty well explained in the tutorials: https://turtlebot3.readthedocs.io/en/latest/sbc_software.html

Where are you stucked? Which model do you have?

KaiqiaoTian, 2017-11-03:
Hi, 
I have turtlebot 3 burger. I follow the tutorials. But I don't know it connect with wifi or bluetooth?

There is no wifi module in turtlebot 3 burger. and I try bluetooth,  it cannot find remote pc.

Thanks
kaiqiao

YuehChuan, 2017-11-03:
@KaiqiaoTian  ou need to have a small router, both remote PC and Raspberry pi3 connect under that router.
ex: 192.168.0. X

routiful, 2017-11-05:
Hi @KaiqiaoTian :)

Raspberry Pi 3 supports wifi and bluetooth. [[Ref](https://www.raspberrypi.org/products/raspberry-pi-3-model-b/)]
So, you can connect between burger and remote pc by using TCP/IP communication.
Before you operate burger, you need to take some steps.
Here is a tutorial link for it.

[[Remote PC]](https://turtlebot3.readthedocs.io/en/latest/pc_software.html#network-configuration)

[[TurtleBot3]](https://turtlebot3.readthedocs.io/en/latest/sbc_software.html#network-configuration-burger-and-waffle)

If you have any problem while you following this step, please feel free to comment for me.
Thanks
Darby

KaiqiaoTian, 2017-11-07:
It works now, under the same router. Thanks guys.

closed 2017-11-07 by routiful