ROBOTIS-GIT/turtlebot3 #89: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver

KaiqiaoTian, 2017-11-07:
Hi, when I running roslaunch turtlebot3_bringup turtlebot3_lidar.launch on Turtlebot3. It shows like that:

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.1.114:11311

core service [/rosout] found
process[turtlebot3_core-1]: started with pid [2100]
 ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/turtlebot3/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
[INFO] [1510071637.611895]: ROS Serial Python Node
[INFO] [1510071637.682395]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1510071639.970637]: Note: publish buffer size is 1024 bytes
[INFO] [1510071639.972320]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1510071639.991698]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1510071640.010225]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1510071640.085019]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1510071640.108005]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1510071645.058791]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1510071645.091485]: Note: subscribe buffer size is 1024 bytes

How can I fix it. Thanks

closed 2017-11-07 by KaiqiaoTian

percymv45, 2018-03-26:
I have a problem similar appear this:

ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]:can´t locate node [hlds_laser_publisher] in package [hls_lfcd_lds_driver]

How can I fix it. Thanks

KaiqiaoTian, 2018-03-26:
You need install lidar driver packages on turtlebot3. Just follow the official instruction. PC and robot has different packages. 
Thanks

percymv45, 2018-03-26:
- I was following the following order
http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#bringup

- although prior to this I had an error when using the following command (cd ~/catkin_ws && catkin_make) my raspberry pi it only loaded up to 92% and the operating system was hung and every time I tried to use the (catkin make) it was hung in that percentage.

routiful, 2018-03-27:
Hi :)

Would you remove `devel` and `build` folder in catkin workspace and try again with below command

`$ cd ~/catkin_ws && catkin_make -j1`

benoit-girard, 2019-07-16:
I have followed the remote PC setup instructions on http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#pc-setup, steps 6.1.1, 6.1.2, 6.1.3.
I did the network configuration (correctly, I hope).
I get the same problem as above, but it comes with an additional problem (the impossibility to open /dev/ttyACM0). Are the two problems linked?

After I entered:
`roslaunch turtlebot3_bringup turtlebot3_robot.launch`

I get:

`started roslaunch server http://10.0.1.52:35461/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://10.0.1.52:11311

process[turtlebot3_core-1]: started with pid [14925]
ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/girard/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
process[turtlebot3_diagnostics-3]: started with pid [14926]
[INFO] [1563288866.638919]: ROS Serial Python Node
[INFO] [1563288866.643528]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1563288866.645354]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
[turtlebot3_core-1] process has finished cleanly
log file: /home/girard/.ros/log/943a9e36-a7d8-11e9-b6c1-080027e160ba/turtlebot3_core-1*.log`

Is it really the problem of not installed lidar packages? I followed your setup instructions, so I suppose it should be there. How can I check that?

benoit-girard, 2019-07-16:
OK, the problem was between the screen and the chair : I did not notice this roslaunch was to be made on the robot, and not the remote PC (it is so obvious when you think about it more than 2 seconds). Sorry again for the inconvenience.