ROBOTIS-GIT/turtlebot3 #91: bad performance in create a map

943230714, 2017-11-28:
After I roslauch turtlebot3_slam turtlebot3_slam.launch, I can see a good performance.But the map was created worse and worse as time went,especially after I turn right or turn left.Is it because my Kinetic does not locate on the center ?or I need to set the parameter of Kinetic?or just because my turtlebot went too slowly?
Thanks a lot

943230714, 2017-12-21:
No, I think it is because the environment in my lab is very similar.
Then it sometime would get a wrong scanmatch.

routiful, 2017-12-21:
Then put certain obstacles(box, bottle,...) in your lab. It makes help to scanmatch.
If it's not an excuse, can you show me a map when you get an error?
It will be more help to me to solve your issue

Thanks
Darby

943230714, 2017-12-21:
I am sorry that I have no time now, But I will show the map when I am free.
Thanks for your help!

robotpilot, 2018-02-12:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-12 by robotpilot