ROBOTIS-GIT/turtlebot3 #92: Create helper script for udev rules installation

tfoote, 2017-11-29:
It's currently done manually in the install instructions: http://turtlebot3.robotis.com/en/latest/sbc_software.html#usb-settings-burger-and-waffle

For the TurtleBot2 we have a simple script that makes it a one liner to setup: https://github.com/yujinrobot/kobuki_core/blob/devel/kobuki_ftdi/scripts/create_udev_rules

This will decrease the likelyhood of user error at this stage of the installation instructions.

tfoote, 2017-11-29:
@allenh1 can you look at doing this?

allenh1, 2017-11-29:
> @allenh1 can you look at doing this?

Absolutely!

tfoote, 2017-12-01:
There appears to be tools to automatically install them using dh_installudev http://manpages.ubuntu.com/manpages/trusty/man1/dh_installudev.1.html

Though there's potentially an issue resolving the `ROS_DISTRO`: 
https://answers.ros.org/question/277124/buildfarm-doesnt-define-the-ros_disto-environment-variable/

A script would still be valuable for from source installations.

routiful, 2017-12-21:
Hello :)

I commit devel branch including script for usb connection.
It will be merged master branch.

Thanks @tfoote and @allenh1 to contribute TurtleBot3 package
Darby

closed 2017-12-21 by routiful