ROBOTIS-GIT/turtlebot3 #94: ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver

rahulrajd, 2017-12-09:
Hi, I have followed all the instructions as per the eManual and also configured the USB permissions. 

aisra@project-aisra:~/catkin_ws$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/aisra/.ros/log/30a683b6-dcd8-11e7-a9a7-9439e5b85887/roslaunch-project-aisra-15135.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.102:36446/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.1.102:11311

core service [/rosout] found
process[turtlebot3_core-1]: started with pid [15153]
ERROR: cannot launch node of type [hls_lfcd_lds_driver/hlds_laser_publisher]: hls_lfcd_lds_driver
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/aisra/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
[INFO] [1512821083.611483]: ROS Serial Python Node
[INFO] [1512821083.620535]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1512821083.624758]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
[turtlebot3_core-1] process has finished cleanly
log file: /home/aisra/.ros/log/30a683b6-dcd8-11e7-a9a7-9439e5b85887/turtlebot3_core-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
How do i fix this issue. 

[image]

routiful, 2017-12-21:
Did you solve this problem??

rahulrajd, 2017-12-21:
Yes. I was running a the file on remote PC. Was so stupid. But later once I
ran it on Turtle bot it worked like a charm

routiful, 2017-12-21:
Great. If you have any problem or question, please feel free to get an issue.

Thanks
Darby.

closed 2017-12-21 by routiful