ROBOTIS-GIT/turtlebot3 #96: How to run in indigo nor kinetic

ncnynl, 2017-12-23:
I want to run in indigo , but all document in kinetic . so  if i want to use indigo , how to change ? or can provide indigo branch to work.

yoshimalucky, 2017-12-26:
Why do you need to use indigo? If you does not have any reason, just update your system to kinetic because the apt packages are not available for indigo.

If you have some serious reasons to use indigo, clone this repository with the other dependent repositories written in package.xml.

ncnynl, 2017-12-31:
@yoshimalucky  Yes,  i use kinetic for TB2 and TB3, the same time use  indigo for TB2 , but some client want to use TB3 for indigo too, them still not update to kinetic . because some software don't support kinetic .

closed 2017-12-31 by ncnynl