ROBOTIS-GIT/turtlebot3 #98: i want to know how to get a distance and angle from Laser Sensor

issong93, 2017-12-30:
Hi, i'm a university student to research in the laboratory from Korea.
I bought  360 Laser Distance Sensor LDS-01 from ROBOTIS company.
I had question about the laser sensor, and i asked to the company. But they want to tell i ask to you.

So, my question is that 10x10 Matrix is shown from matlab/simulink through the Sensor.
But i want to know a distance and angle from the Sensor. I don't know understand the 10x10 Matrix's mean. Please can you teach me about solution as soon as possibe. Thank you.

[image]

routiful, 2018-01-02:
Hi :)

Can you explain how to get this variable on MATLAB? 

Thanks
Darby

issong93, 2018-01-02:
I just received the data of lidar using Serial port of Matlab/Simulink.

[image]

routiful, 2018-01-02:
Before getting data from lidar, you need to communicate with it.

[lidar comm example](https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver/blob/master/src/hlds_laser_publisher.cpp)

Above code is example to show how to begin lidar sensor.

Thanks
Darby

issong93, 2018-01-06:
Really thanks for your reply

I will try that code

issong93, 2018-01-08:
Where do i get the header files?

#include <ros/ros.h> 
35 #include <sensor_msgs/LaserScan.h> 
36 #include <boost/asio.hpp> 
37 #include <hls_lfcd_lds_driver/lfcd_laser.h> 
38 #include <std_msgs/UInt16.h>

robotpilot, 2018-02-12:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-12 by robotpilot