ROBOTIS-GIT/turtlebot3 #99: Turtlebot3 wrong position Lidar in Rviz

patrick95470, 2018-01-01:
HI,

First of all happy new year to all of you, Hope this new year will bring you everything you are dreaming of

I'm a rookie , I just finalized the mounting of the Turtlebot3, looks amazing , and made my first teleoperation with success.
But My concern is the lidar and Rviz, when I approach object in direction to the turtlebot, on Rviz, the visualization is not matching with the reality ( wrong positioning of the points vs where they are in real  ) 

Would it be possible to help me on this issue

Kind regards

routiful, 2018-01-02:
Hello @patrick95470  :) Happy new year!!

Did you check '/scan' topic message? It will be help you to watch the invalid data in detailed

Thanks
Darby

patrick95470, 2018-01-02:
Hi Darby,

Many thx for your swift reply, but could you please be more specific in what I have to do ?.. ;-)

Regards

Patrick

routiful, 2018-01-04:
I mean check a scan data using topic message. /scan shows range(distance) center of sensor to object.
You can check it using 'rqt'

Thanks
Darby

patrick95470, 2018-01-17:
Hi Darby

I have to say that my skill is not enough high to see exactly what I have to do, I have installed rqt, I hope it has been done correctly, ( how to check??) 
but afterward, I'm puzzled in the way to do checking you suggest me
Would it be possible to write shortly the commands I have to use..;

thx

Kind regards

Patrick

PandaJules, 2018-01-18:
Hi Patrik,

I'm also using Turtlebot3 with ROS, so perhaps I can help. So suppose you have teleop running - that will be in one terminal. Open another terminal (or a tab within your terminal with Ctrl+Shift+T) and type `rostopic list`. You'll see all the topics that are being published/subscribed to. If you don't know about topics, read here http://wiki.ros.org/Topics. In the list, there should be one topic called **/scan**. This is how you make Turtlebot3 use lidar data by subscribing to it and reading the messages that are published there. To see what kind of messages are used on that topic type `rostopic info /scan` and it will say **sensor_msgs/LaserScan*. Now what on earth is that? Type `rosmsg show sensor_msgs/LaserScan` and you'll see the structure of the message. The array of ranges is what you are after. Final part, to actually see what's happening on that topic type `rostopic echo /scan`. That will give you a gazillion messages per second, so you won't be able to analyse them on the go. You are better off stopping this by Ctrl+C and going up, looking for the work **ranges** and seeing the values. It will be an array of 360 values (one for each degree) and it will contain values in meters for what the lidar thinks is ahead of you.

patrick95470, 2018-01-19:
HI PandaJules

I will follow carefully what you advised me and let you know

Did you get the same concern with your TB3?

In fact I can teleop run it without problem, the TB3 moves in the right direction but the position in RVIZ is not correct, there is a shift between the reality and the points of the lidar's representation

Kind regards

PAtrick

PandaJules, 2018-01-19:
One thing you need to remember is that lidar has minimum and maximum range which I think are 0.12 and 3.5 meters. So, for example if you come to a wall very close, you can get this weird discontinuity.
[image] 

If the walls are circular, then maybe the ray didn't come back to the lidar, so it thinks there is nothing. I attach 2 screenshots of the cloud point that I usually see. I don't use RViz very often, to be honest. 
[image]

patrick95470, 2018-01-19:
HI

in fact I have the clouds points but the orientation of the walls does not stick with the position of TB3 on RVIZ and the reality

I will copy/past a screenshot tomorrow to explain my words

kind regards

patrick95470, 2018-01-20:
Hi
The TB3 is a middle of the room, but this is what I get ib RVIZ

[image]

in the fact the TB3 is not well positioned in relation to its real environment in RVIZ

kind regards

patrick95470, 2018-01-20:
My TB3 in RVIZ is not well positioned on the Y axes

If somebody can tell me how to fix this issue it would be great

PandaJules, 2018-01-22:
Is the "rainbow line" following a wall? What I mean is - are the points _mostly_ correct, but you are unsure about those points in the middle of nowhere? It might be noise or you have shiny objects that react weirdly with the rays. Try positioning the turtlebot in a place with good vertical walls that make up a square a see what RVIZ thinks

patrick95470, 2018-01-22:
The Lidar is working properly, the walls are well defined

but my concern is that in RVIZ the TB3 is not at the position where it should be, in my screen copy, the TB3 is far away of the wall, but in RVIZ it seems to touch the wall....

routiful, 2018-01-23:
How do you open Rviz?
Did you follow this [tutorial](http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#turtlebot3-burger)?

patrick95470, 2018-01-23:
Hi
I did it, but I have a doubt concerning the commands, I just typed : 

$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

What about : 
$ roslaunch turtlebot3_bringup turtlebot3_lidar.launch
$ roslaunch turtlebot3_bringup turtlebot3_core.launch

should I type them too ?

routiful, 2018-01-23:
No, turtlebot3_robot.launch includes lidar and core file.

patrick95470, 2018-01-23:
hummm, so I'm doing exactly what we have in the tutorial ....

As I wrote, I can teleoperate the TB3 without problem, it is just the positioning into RVIZ which is not correct, there is a shift , the TB3 is not moving is the right direction , 

We have all the wall clearly visible in RVIZ, but if I move it for instance right away out of the wall, in RVIZ we see the TB3 moving toward to the wall.. :-(

jhausladen, 2018-01-23:
Hi,

we think we have a similar problem with our TurtleBot3. Generally, driving in teleop mode works fine. Also the lidar seems to work properly when creating maps using the SLAM tutorial.

However when we try to navigate on the map, the robot just spins in a circle.

First the robot was positioned somewhere on the map:

[image]

When pressing the "2D Pose Estimate" button, TurtleBot3 finds the right position in the room and highlights the walls correctly:

[image]

But when we try to navigate somewhere on the map, it just begins to spin around and location information in RViz is displaced.

[image]

According to the tf error shown in RViz, we checked the time on both machines but it was synced correctly.

patrick95470, 2018-01-23:
Hi

I would say that I’m not feeling alone right now, but the question is how to fix this issue

If somebody could help us it would be great

jhausladen, 2018-01-24:
Yesterday we also set up our waffle with a Raspberry Pi instead of the Intel Joule and suddenly navigation worked. Walls are now mapped correctly in RViz and TurtleBot3 doesn’t spin as hell as before. In our case it seems that the Intel Joule Ubuntu Distro, or especially some  of the ROS  packages cause that issue.

patrick95470, 2018-01-29:
Hey

thx for this important information, indeed I guess that many update fixes had been done, now works  :-)

closed 2018-01-29 by patrick95470