ROBOTIS-GIT/turtlebot3 #101: ros2

unverciftci, 2018-01-08:
Do you plan to support ROS2?

routiful, 2018-01-23:
Thank you for your inquiry!

Understandly, we have a plan to support ROS2 near future. But we don't have any details.

Thanks :)
Darby

robotpilot, 2018-02-12:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

minggangw, 2018-04-12:
any progress on the ROS2 support? thanks!

minggangw, 2018-04-13:
Hi @robotpilot, thanks for your quick and useful update about the progress of support ROS2 on TurtleBot3.

We are developing a Node.js client of ROS2 - [rclnodejs](https://github.com/RobotWebTools/rclnodejs), and we want to promote the web solution on ROS2 plaform (We are collaborating with [Web Robot Tools](https://github.com/RobotWebTools) and have implemented some modules). I understand that ROS2 is at the early stage, but we want to deploy and have fascinating demo on TurbleBot, which has been used as a prototype widely. So we are looking forward to the support of it.

Thanks for your hard work :)

robotpilot, 2018-05-27:
Hi @minggangw,
We also want to quickly see how you work with your development Node.js. 
We will try our best. Thank you. :)

minggangw, 2018-05-28:
You are welcome to try the Node.js [client](https://github.com/RobotWebTools/rclnodejs) for ROS2, and I think we can collaborate together to develop some demos by using JavaScript if possible, thanks!

OpusK, 2018-08-29:
Hi, all.

Currently, we are working on ROS2 support for TB3.
Some features (teleop test available) are implemented and will be released in a test version soon.

For more information, please refer to the following repository.
- [TurtleBot3 ros2](https://github.com/ROBOTIS-GIT/turtlebot3/tree/ros2)
- [turtlebot3_msgs for ros2](https://github.com/ROBOTIS-GIT/turtlebot3_msgs/tree/ros2)
- [OpenCR for ros2-micrortps](https://github.com/ROBOTIS-GIT/OpenCR/tree/feature-ros2-micrortps)

minggangw, 2018-08-29:
Glad to hear the progress, thanks for your  dedicated work!

robotpilot, 2018-09-05:
We have released a tutorial on applying ROS2 to TurtleBot3.
https://discourse.ros.org/t/tb3-introducing-ros2-tutorials/5959
http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#ros2

So I will close this issue. If you are having trouble using ROS2 with TurtleBot3, please open a new issue at any time. We will support you. :)

closed 2018-09-05 by robotpilot