ROBOTIS-GIT/turtlebot3 #104: how to test the openCR card ( to see in all sensors are working properly)

patrick95470, 2018-01-17:
Hi
I have followed the tutorial from the robotis emanual by usng this command : 
rosrun rosserial_python serial_node.py __name:=turtlebot3_core _port:=/dev/ttyACM0 _baud:=115200

but this is what I'm getting : 

~$ rosrun rosserial_python serial_node.py __name:=turtlebot3_core _port:=/dev/ttyACM0 _baud:=115200
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/rosserial_python/serial_node.py", line 48, in <module>
    rospy.init_node("serial_node")
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 326, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 514, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1053, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 897, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 859, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 836, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
socket.error: [Errno 111] Connection refused

If someone could help me it would be helpfull

kind regards

patrick

routiful, 2018-01-19:
Hi :)

Did you set a USB port?
Please following this [link](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr1_0_software_setup/#usb-port-settings). 

Thanks
Darby

patrick95470, 2018-01-19:
Hi Darby

many thx I guess this is the issue

kind regards

patrick

closed 2018-01-19 by patrick95470

unverciftci, 2018-04-30:
the link is not working

routiful, 2018-05-02:
@unverciftci 

That link was broken. I give you new link

http://emanual.robotis.com/docs/en/parts/controller/opencr10/#usb-port-settings