ROBOTIS-GIT/turtlebot3 #106: Lidar in Gazebo NOISE simulation

PandaJules, 2018-01-18:
I've been trying to analyze deviation in ranges that I read from /scan topic while Turtlebot3 (burger) is stationary, but then I realised I should just look in models and this is what I found in `turtlebot3_burger.gazebo.xacro` (omitting some detail)

`<gazebo>
    <plugin name="imu_plugin" filename="libgazebo_ros_imu.so">
      ...
      <gaussianNoise>0.0</gaussianNoise>
      <imu>
        <noise>
          <type>gaussian</type>
          <rate>
            <mean>0.0</mean>
            <stddev>2e-4</stddev>
            <bias_mean>0.0000075</bias_mean>
            <bias_stddev>0.0000008</bias_stddev>
          </rate>
        </noise>
      </imu>
    </plugin>
  </gazebo>
`
and then in another place

` <gazebo reference="base_scan">
    <sensor type="ray" name="lds_lfcd_sensor">
      <ray>
        <scan>...</scan>
        <range>... </range>
        <noise>
          <type>gaussian</type>
          <mean>0.0</mean>
          <stddev>0.01</stddev>
        </noise>
</ray>
`

So I'm a bit confused about which Gaussian noise you are modeling (which mean/std) and why these particular values? I want to later on move on from simulation to using the actual TB3 and I would like to use realistic values.

routiful, 2018-01-19:
Hi :)

The gaussian noise is optional. I purposed to inform that just simple code applies noise.
If you don't need it, edit the code yourself as your development environments.

Best regards
Darby

PandaJules, 2018-01-19:
Thank you for the insight, I'll try playing around with the values. 
I was just wondering to check how it corresponds to the real TB, so that when I move on from simulation, things will be similar. For example, gazebo says that wheel radius is 0.03, whereas it should be 0.033 or wheel separation is 0.14 and it should be 0.16. Same goes to the maximum speed of the robot, as it says in the specification - 0.22m/s for the burger, but in urdf it's 0.1 for both wheels (but the simulation works fine when they move faster...) Not sure how gazebo extracts the information from this xacro.

routiful, 2018-01-22:
Hi :)

I set a information about TB3 burger on [turtelbot3_burger.gazebo.xacro](https://github.com/ROBOTIS-GIT/turtlebot3/blob/master/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro) file.

Would you explain to me where do you get the wrong information?? then i will check it.

Thanks 
Darby

PandaJules, 2018-01-30:
For example, `<maxVel>0.1</maxVel>` for right and left wheels, but in specifications for Turtlebot it says that maximum vel is 0.22 m/s. And then these values in xacro file do not prevent Gazebo from making TB run at VERY HIGH speeds that would be unrealistic in real life. I just want to use the code later on without breaking the real robot. I wonder how Gazebo extracts xacro information, because for wheels, even though you set ` <wheelSeparation>0.160</wheelSeparation>` and `<wheelDiameter>0.066</wheelDiameter>`, it still gives 0.14 and 0.06 which messes up some calculations.

routiful, 2018-02-01:
Hi :)

Your right. <maxVel> is not corresponds to real TB3. I will modify it or if you want, you can change it and PR to develop branch.

But i still wonder how to check 'wheelSeparation' value in Gazebo. Please tell me where you get a information of it.

Best regards
Darby

robotpilot, 2018-02-13:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-13 by robotpilot