ROBOTIS-GIT/turtlebot3 #108: checksum error when launching turtlebot3_bringup

yamsam, 2018-01-21:
Hi,
I have been getting the following error messages every time when i launched turtlebot3 on my burger since i updated the source from the current master yesterday.

Do you have any advice how to fix this issue? 

```

isamu@isamu-desktop:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch 
... logging to /home/isamu/.ros/log/a7d79cc0-fe63-11e7-bccf-bc5ff4f9cb8e/roslaunch-isamu-desktop-3246.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.22:38322/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.21:11311

process[turtlebot3_core-1]: started with pid [3255]
process[turtlebot3_lds-2]: started with pid [3256]
process[turtlebot3_diagnostics-3]: started with pid [3257]
[INFO] [1516508967.049772]: ROS Serial Python Node
[INFO] [1516508967.083459]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1516508969.227216]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c
[INFO] [1516508969.240459]: Note: publish buffer size is 1024 bytes
[INFO] [1516508969.241765]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1516508969.252847]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1516508969.274881]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1516508969.293139]: Setup publisher on joint_states [sensor_msgs/JointState]
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[INFO] [1516508970.359227]: Setup publisher on /tf [tf/tfMessage]
[INFO] [1516508970.377699]: Note: subscribe buffer size is 1024 bytes
[INFO] [1516508970.379390]: Setup subscriber on cmd_vel [geometry_msgs/Twist]
[INFO] [1516508970.382394]: Start Calibration of Gyro
[INFO] [1516508971.754625]: Calibrattion End
[ERROR] [1516508971.763685]: Tried to publish before configured, topic id 125
[ERROR] [1516508971.777498]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c
[INFO] [1516508971.782853]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1516508971.788025]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1516508971.793043]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1516508971.798474]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1516508971.803801]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1516508971.835651]: Tried to publish before configured, topic id 125
[ERROR] [1516508971.842810]: Tried to publish before configured, topic id 125
[ERROR] [1516508971.860354]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c
[INFO] [1516508971.864091]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1516508971.867687]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1516508971.871083]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1516508971.874550]: Setup publisher on joint_states [sensor_msgs/JointState]
[INFO] [1516508971.878085]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1516508971.888718]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c
`
```

routiful, 2018-01-22:
Hi :)

You have to update turtlebot3_msgs and firmware to use new version of turtlebot3.
This [instruction](https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729) will help you .

Thanks.
Darby

yamsam, 2018-01-22:
Hi Darby,
Thank you for your advice.

After updating new firmware it worked without error again!

closed 2018-01-22 by yamsam

damianangelo1712, 2018-06-20:
Hello,
  I am getting the same error as above. I am not using Arduino, but raspberry PI for my turtle bot (waffle pi). I have updated turtbot3_msgs multiple times but I receive the saame check sum error. I have manually installed each ROS package but still no developments. I appreciate any help regarding this matter.
Regards,
Damian
PS: Please bear with me since I am totally new to this area

routiful, 2018-06-21:
Hi @damianangelo1712  :)

You have to upload firmware to OpenCR and download software through github.
e-Manual is the best reference to solve it and refer below link

- [How to setup](http://emanual.robotis.com/docs/en/platform/turtlebot3/setup/#setup)

- [Discourse](https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888/1)

Thanks

routiful, 2018-06-22:
Your problem is caused by not matching topics between `turtlebot3_msgs` and `turtlebot3_ros_lib` in OpenCR. You don't need to install Arduino IDE. 
After delete all related turtlebot3 packages and clean `catkin_workspace`, you may follow setup reference again.

Thanks

damianangelo1712, 2018-06-27:
Thanks for your help mate. I did what you suggested and sourced the bashrc file. Now i am getting the following error. Ping from Master to Slave and vice-versa works fine. ANy inputs are appreciated...
[image]

damianangelo1712, 2018-06-28:
> Hello @damianangelo1712 <https://github.com/damianangelo1712> :)
>
> roscore is needed once. You can find reference I attached below
>
> http://emanual.robotis.com/docs/en/platform/turtlebot3/
> bringup/#run-roscore
>
> Thanks

Hey mate,
  I get that part. I am getting error during roslaunch bring up as well. I
added that image just to show you that. Rosrun and Roslaunch works properly
at the master but not at the slave. When i run Rviz the wheel left and
wheel right show no transform from base.
Do you know about these errors?

Best regards,
Damian

routiful, 2018-06-29:
Sorry. I can't understand your problem. Can you show me more information as Rviz images or error messages?