ROBOTIS-GIT/turtlebot3 #110: Adding dynamixel to TurtleBot3 while OpenCR board is in Turtlebot3_core configuration

newrobofreak, 2018-01-22:
Hello,
I carefully searched for this topic, but the only one I found that [wrote about it ](https://github.com/ROBOTIS-GIT/turtlebot3/issues/90) never answered back. 
If I am right, the examples provided for _DynamixelSDK_ are designed for the use with OpenCR1.0 board configured as USB2Dynamixel adapter. However, in my project we need to control further Dynamixels (ID 3, ID 4) while the OpenCR board is in _Turtlebot3_core_ configuration.

Since I am completely new to this topic, I hope you can give me the right direction.

I thought about writing a class like _Turtlebot3MotorDriver_, include its functionality in the  _Turtlebot3_core.ino_ and upload this with _ArduinoIDE_. If I am right, most functions used in _Turtlebot3MotorDriver_ are based on the _DynamixelSDK.h_, so I would have to make myself familiar with the _dynamixel-sdk_, right? But the _dynamixel-workbench_ package would not be relevant for me in this case, right? 
This idea seems very difficult to me though, is there an easier way?
Are there any examples in this direction?

Thanks a lot in advance.

newrobofreak, 2018-01-23:
Thanks a lot Darby, I did not see this example before. It is definitely the right direction.
I do not plan to make a four-wheel drive though. I plan to use the TurtleBot3 as delivered with two Dynamixels for driving, laser-scanner and everything and just add the two Dynamixels for manipulation. Those two should later only rotate +/-45 degree in two joints of a manipulator.
What I would need is something like the example, where the Open Manipulator is mounted on top.  I could not find such an example for the ArduinoIDE under _~/.arduino15/packages/OpenCR/hardware/OpenCR/1.0.15/libraries/turtlebot3/examples/turtlebot3_friends_ though. Under the [open_manipulator page](https://github.com/ROBOTIS-GIT/open_manipulator) there is a folder for TurtleBot3. But this does not include the OpenCR1.0 Arduino software, right? Or did I miss something?
You are still working on this software if I read the e-Manual right. Is the above mentioned way with the Turtlebot3MotorDriver promising, or what would be your advise?
Best regards :)
Johannes

routiful, 2018-01-25:
Hello :)

Thank you for your inquiry about OpenManipulator.
Yes, I am working to develop library for OpenManipulator. But it is not included in OpenCR example yet.

Additionally, you can attach Dynamixel on Waffle-plate.

[image]
 
Thanks 
Darby

newrobofreak, 2018-02-02:
Thanks a lot. Nice to know that you can connect them here as well. 
Modifying the files from the original Turtlebot core software worked fine for me (Turtlebot3_core.ino, Turtlebot3_core_config.h and Turtlebot3_motor_driver.h, Turtlebot3_motor_driver.cpp). I added Dynamixel with ID 3 and 4 and implemented a new group_sync_write for writing goal position to them.
So I think, I can close this question for now.
Thanks a lot for the fast answers again :)

closed 2018-02-02 by newrobofreak

tedalias, 2018-06-11:
Hi @newrobofreak :)
I am currently having kind of the same issue here, I just want to add a dynamixel as a manipulator to the TB3.
Could you tell me roughly what changes you did to the mentioned files?
Best regards:)

stevensu1838, 2020-09-10:
> Thanks a lot. Nice to know that you can connect them here as well.
> Modifying the files from the original Turtlebot core software worked fine for me (Turtlebot3_core.ino, Turtlebot3_core_config.h and [...]

> Thanks a lot. Nice to know that you can connect them here as well.
> Modifying the files from the original Turtlebot core software worked fine for me (Turtlebot3_core.ino, Turtlebot3_core_config.h and [...]

Hey buddy,
Could you please share your customized code with me? Thanks a lot.  email:stevensu1838@gmail.com