ROBOTIS-GIT/turtlebot3 #111: How can i make return system?

par8057, 2018-01-24:
Hi im working with turtlebot3

I want to make literally 'return system' which means that Turtlebot3 goes to goal position and return to the
initial position.
I already succeed to run turtlebot3_navigation package.
But i dont know how to edit a package to make 'return system'
Do you know what i mean???

Please let me know guys

routiful, 2018-01-25:
Hello :)

I can understand what you mean. 
I expect [turtlebot3_deliver package](https://github.com/ROBOTIS-GIT/turtlebot3_deliver) can help your project.

Best regards
Darby

robotpilot, 2018-02-12:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-12 by robotpilot