ROBOTIS-GIT/turtlebot3 #113: Turtlebot3 Bringup Error

pesal80, 2018-01-25:
Hi,
I tried to bring up the turtlebot3 with command below:
"roslaunch turtlebot3_bringup turtlebot3_robot.launch"

However, it seems it doesn't work properly. Errors show up:
[INFO] [1516852585.597507]: Setup publisher on /tf [tf/tfMessage]
[ERROR] [1516852026.681106]: Tried to publish before configured, topic id 125
[ERROR] [1516852026.685209]: Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c

I tried to catkin_make the turtlebot3 packages few times but nothing improve.
I still manage to control the robot via teleop. However, when I run rviz, nothing is shown in the rviz including the robot model. Error I got in the rviz:
1. Parameter [robot_description] does not exist, and was not found by searchParam()
2. For frame [base_scan]: Frame [base_scan] does not exist

Could anyone help me to solve the problem?

routiful, 2018-01-25:
Hi :)

I think you did not update Firmware. 
Please check this link. 

[TB3 version update](https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729)

Thanks
Darby

pesal80, 2018-01-25:
Hi Darby,
Aha, that makes sense. I skip the steps for OpenCR Firmware update.
Thanks for your prompt reply. I will try it.

Cheers

mw-sandipk, 2018-02-09:
Updating firmware is failing, and would appreciate any suggestions: https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729/17?u=tosandip

robotpilot, 2018-02-12:
Hi @mw-sandipk ,
Has the issue been resolved?

mw-sandipk, 2018-02-12:
@robotpilot the issue has not been resolved. Can you please suggest what would be a goo way of debugging/ going forward?

routiful, 2018-02-13:
Hi @mw-sandipk 

This TB3 update does not need to update bootloader, you could use it to update firmware.
[Update firmware by board manager](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-the-opencr-package-via-boards-manager)

But if you want to burn bootloader, you have to reset opencr with pushed boot button.
Please check this [manual](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#update-bootloaderlinux)

Thanks
Darby

mw-sandipk, 2018-02-13:
@routiful I think am misunderstanding something, so when you informed about the big update: https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729, and then had "How to update" (1) Software (2) Firmware

Did you mean the (2) firmware update steps to be done TB3 Linux installed? I.e., the Arduino IDE is installed on the tb3-waffle ? and not on a remote desktop connected to openCR board on the waffle?

What I have been trying to do so far is, I have connected the OpenCR to a desktop using USB port and the Arduino IDE was installed on the desktop -> through which I was trying to update the firmware on openCR -> hence burn bootloader felt like that right thing to do to update anything on the openCR.

What I was doing  -> is that another way of updating the firmware? During all this, I was putting the openCR board in DFU mode and then trying to burn bootloader- > which is failing.

What do you recommend?

mw-sandipk, 2018-02-15:
@routiful can you suggest if what am doing is correct OR I should do everything on the Linux in Turtlebot?

mw-sandipk, 2018-02-16:
I am bumping into another problem, after updating the firmware, even the basic operations of moving the robot forward/rotate (using these http://emanual.robotis.com/docs/en/platform/turtlebot3/hardware_setup/#basic-operation) are not working any more.

Just to re-iterate I did the step 7.6.2 on this page: http://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-the-opencr-package-via-boards-manager

What is going one?

routiful, 2018-02-19:
Hello @mw-sandipk :)

Would you inform to me that which version of turtlebot3_core?

Before you operating TB3, you can check actuators using turtlebot3_setup_motor (Open Arduino IDE -> File -> Examples -> TurtleBot3 -> turtlebot3_setup -> turtlebot3_setup_motor)

Thanks
Darby

mw-sandipk, 2018-02-23:
What do you mean by turtlebot3_core? I looked at the launch file:

```
turtlebot3@turtlebot3:~$ roscd turtlebot3_bringup
turtlebot3@turtlebot3:~/catkin_ws/src/turtlebot3/turtlebot3_bringup$ cd launch
turtlebot3@turtlebot3:~/catkin_ws/src/turtlebot3/turtlebot3_bringup/launch$ cat turtlebot3_core.launch 
<launch>

  <node pkg="rosserial_python" type="serial_node.py" name="turtlebot3_core" output="screen">
    <param name="port" value="/dev/ttyACM0"/>
    <param name="baud" value="115200"/>
  </node>

</launch>
turtlebot3@turtlebot3:~/catkin_ws/src/turtlebot3/turtlebot3_bringup/launch$
```

I also ran the turtlebot3_core

```
turtlebot3@turtlebot3:~$ roslaunch turtlebot3_bringup turtlebot3_core.launch 
... logging to /home/turtlebot3/.ros/log/c413415a-18e4-11e8-a0e4-a0c589238c19/roslaunch-turtlebot3-2411.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.31.28.61:45717/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

auto-starting new master
process[master]: started with pid [2423]
ROS_MASTER_URI=http://172.31.28.61:11311

setting /run_id to c413415a-18e4-11e8-a0e4-a0c589238c19
process[rosout-1]: started with pid [2436]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [2439]
[INFO] [1519423147.564850]: ROS Serial Python Node
[INFO] [1519423147.579496]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1519423164.688909]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
^C[turtlebot3_core-2] killing on exit
[INFO] [1519423168.528072]: Send tx stop request
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I did upload the turtlebot3_setup_motor and in the end got this:

```
opencr_ld ver 1.0.2
opencr_ld_main 
>>
file name : /tmp/arduino_build_708293/turtlebot3_setup_motor.ino.bin 
file size : 113 KB
Open port OK
Clear Buffer Start
Clear Buffer End
>>
Board Name : OpenCR R1.0
Board Ver  : 0x17020800
Board Rev  : 0x00000000
>>
flash_erase : 0 : 1.126000 sec
flash_write : 0 : 1.249000 sec 
CRC OK B2E1F0 B2E1F0 0.004000 sec
[OK] Download 
jump_to_fw 
```

routiful, 2018-02-27:
You can check firmware version in turtlebot3_core_config.h in Arduino.
And After upload turtlebot3_setup_motor.ino, then you have to open `serial monitor` in Arduino

routiful, 2018-03-09:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-09 by routiful