ROBOTIS-GIT/turtlebot3 #118: Install Dependent Packages issue

patrick95470, 2018-01-29:
Hi

Due to uncorrect working of my TB3, I decided to remake from scratch  the installation .

thus after having installed ubuntu, installed ROS without problem

now I have a huge concern when it is time to Install Dependent Packages

the RPI freezes when I make the command  cd ~/catkin_ws && catkin_make

it stopsall the time at 94% like the picture is showing

if somebody has an idea of what's going wrong ?

[image]

patrick95470, 2018-01-29:
I just tried to delete all folders inside  catkin_ws/src

and retype all commands , but now the RPI stops at 64% !!!!???? :-(
[image]

patrick95470, 2018-01-29:
And redone the same, now it is stopping at 94%.....

robotpilot, 2018-01-29:
Hi @patrick95470 
In your case, It would be nice to use the following command.

`cd ~/catkin_ws && catkin_make -j1`

patrick95470, 2018-01-29:
Ok thx, works now

Can you please tell me more about this attribute ?

robotpilot, 2018-01-29:
@patrick95470 
This is one option of the 'make' command:
"-j [N], --jobs [= N] Allow N jobs at once; infinite jobs with no arg". 

If you set the option to '-j1', only one process will be used when building. 

P.S. Building all of the TurtleBot packages on SBC(Raspberry Pi 3) seems pretty heavy. So, we will fix it soon. Thanks.

closed 2018-01-29 by patrick95470