ROBOTIS-GIT/turtlebot3 #125: have you experienced a crash with teleop

patrick95470, 2018-02-02:
Hi

I tried to run the teleop with a XBOX360 joystick, but when I launch 
roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

it crashes 
[image]

if somebody has an idea of what is going wrong

:-)

routiful, 2018-02-05:
Hi :)

Did you try below reference?
[xbox-360-joystick](http://emanual.robotis.com/docs/en/platform/turtlebot3/teleoperation/#xbox-360-joystick)

patrick95470, 2018-02-05:
Hi

I did it and I have this error ...

patrick95470, 2018-02-05:
[image]

routiful, 2018-02-06:
Hi :)

Have you used keyboard to control turtlebot3 with teleop_twist_keyboard
pkg??
Does it shows same error window?

-- 

Taehoon Lim (thlim@robotis.com)
Researcher / R&D Dept. OpenSourceTeam

ROBOTIS CO., LTD. (http://www.robotis.com)
#1505 Ace High End Tower NO. 3,
Gasan-dong 145, GasanDigital 1-ro,
GeumCheon-gu, Seoul, Korea 153-787
Tel : +82-70-8671-3380, Fax: +82-70-8230-xxxx

Follow ROBOTIS on Twitter <https://twitter.com/robotis>, Facebook
<https://www.facebook.com/robotis.company>, Linkedin
<http://www.linkedin.com/company/robotis>

patrick95470, 2018-02-06:
Hi

it works fine with the keyboard, no errors and I can move the TB3, that's only with the xbox360 joystick that I got the issue ..

robotpilot, 2018-02-13:
Hi @patrick95470 ,
I hope your work goes well. I think this is a driver issue or a problem with the package, regardless of TurtleBot3. Thank you and please ask if you have further questions. This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-13 by robotpilot