ROBOTIS-GIT/turtlebot3 #126: multi turtlebot waffle

nipkor, 2018-02-07:
Hi guys ...
i wanted to know that is it possible to run multiple turtlebot3 waffle on one rosmaster?

I tried with using tf_prefix and namespaces and every thing went good... But there is a problem.
the robot position transform message (base_footprint) published by turtlebot3_core does not match the tf_prefix .
 in other words i want the turtlebot3_core to publish something like blabla/base_footprint instead of base_footprint

Thank you all ,prior to your assistance guys 

@ROBOTIS-OpenSourceTeam

robotpilot, 2018-02-12:
Yes. no problem. :)

Will you use multiple robots like the video link below? 
We used three TurtleBot3 under the name 'Delivery Service Robot' in the ICRA2017 demonstration.
Videos: https://youtu.be/5NYdgnzxykc and https://youtu.be/u7qb8xiSiNg

We describe this in detail in Chapter 13 of the book link below. You can refer to it.
http://community.robotsource.org/t/download-the-ros-robot-programming-book-for-free/51/

robotpilot, 2018-02-13:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-02-13 by robotpilot