ROBOTIS-GIT/turtlebot3 #127: Update cause Raspberry Pi hang

krizchong, 2018-02-08:
After i updated Turtlebot folder in my Raspberry Pi of TB3 Burger according to https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729, then when I try to catkin_make, Pi will just hang as
[image]

Then I remove turtlebot3_patrol folder and catkin_make again, Pi always hang at
[image]

Pi will just hang and even the mouse pointer cannot move at all!
What was happened? Any advice?

kijongGil, 2018-02-08:
Hi :)

I think that this problem is error about multi-core build.

Try catkin_make again with single-core.

**catkin_make -j1**

Thanks
Gilbert

krizchong, 2018-02-08:
Thanks Gilbert for your reply.

Justnow I remove the Turtlebot3 folder then clone again from github, then it can compile without freezing.

I will try catkin_make -j1 next time if facing similar problem. Thank you ^^

closed 2018-02-08 by krizchong