ROBOTIS-GIT/turtlebot3 #128: No motor setup instruction for replacement

yoshimalucky, 2018-02-08:
Following the emanual, I tried to build my turtlebot waffle from the bulk supplies.

However, it did not work. Then, I figured out that I had to run the motor setup program "turtlebot3_setup" from arduino IDE. Then, I noticed that i had to set up the motors by sending some commands from the console in the IDE.

Finally, it worked but spent a lot of time. We need to add the content of how to set up the new motor for replacement to the emanual.

robotpilot, 2018-02-12:
@yoshimalucky 
Thank you for your feedback. :)
We will make a guide for what you say. This section now requires a lot of information about OpenCR.
So we describe this in detail in Chapter 9 '9.3.3. TurtleBot3 Setup Firmware' of the book link below. You can refer to it.
http://community.robotsource.org/t/download-the-ros-robot-programming-book-for-free/51/
And, we will make some excerpts and upload them to the e-manual.

routiful, 2018-03-09:
We add '[How to setup Dynamixel for TurtleBot3](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#turtlebot3-setup-dynamixels)' in WIKI.

Please following above link :)

closed 2018-03-09 by routiful