ROBOTIS-GIT/turtlebot3 #129: cMake Error

wizard007, 2018-02-13:
Dear Robotis,

Pardon if this issues has been raised before.
I just purchased TB3 Burger. My Pie3 works just fine but I was annoyed how slow Pi3 was so I purchased the RealSense Dev Kit with UP board.

All installations for RealSense works fine however, following all instructions from emanual.robotis.com. I had issues with the Cmake: (bare in mind i am complete newbie, so pardon in adv)

`syawal@syawalUP:~$ cd ~/catkin_ws && catkin_make
Base path: /home/syawal/catkin_ws
Source space: /home/syawal/catkin_ws/src
Build space: /home/syawal/catkin_ws/build
Devel space: /home/syawal/catkin_ws/devel
Install space: /home/syawal/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/syawal/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/syawal/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/syawal/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - turtlebot3 (metapackage)
-- ~~  - turtlebot3_msgs
-- ~~  - hls_lfcd_lds_driver
-- ~~  - master_bridge
-- ~~  - cvra_operations
-- ~~  - turtlebot3_bringup
-- ~~  - turtlebot3_interactions
-- ~~  - turtlebot3_lds
-- ~~  - turtlebot3_patrol
-- ~~  - turtlebot3_pointop
-- ~~  - debra_description
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'turtlebot3'
-- ==> add_subdirectory(turtlebot3/turtlebot3)
-- +++ processing catkin package: 'turtlebot3_msgs'
-- ==> add_subdirectory(turtlebot3_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- turtlebot3_msgs: 4 messages, 2 services
-- +++ processing catkin package: 'hls_lfcd_lds_driver'
-- ==> add_subdirectory(hls_lfcd_lds_driver)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
-- +++ processing catkin package: 'master_bridge'
-- ==> add_subdirectory(cvra_debra_ros/master_bridge)
-- +++ processing catkin package: 'cvra_operations'
-- ==> add_subdirectory(cvra_debra_ros/cvra_operations)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "sick_tim" with any
  of the following names:

    sick_timConfig.cmake
    sick_tim-config.cmake

  Add the installation prefix of "sick_tim" to CMAKE_PREFIX_PATH or set
  "sick_tim_DIR" to a directory containing one of the above files.  If
  "sick_tim" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  cvra_debra_ros/cvra_operations/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'sick_tim'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "sick_tim" with any
  of the following names:

    sick_timConfig.cmake
    sick_tim-config.cmake

  Add the installation prefix of "sick_tim" to CMAKE_PREFIX_PATH or set
  "sick_tim_DIR" to a directory containing one of the above files.  If
  "sick_tim" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  cvra_debra_ros/cvra_operations/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/syawal/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/syawal/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:290: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

`

robotpilot, 2018-02-13:
Hi @wizard007 

This error is not an error related to TurtleBot3.
Your error message says that there is no sick-tim package. Are you using a SICK sensor?
If so, install it with the following command to resolve the problem.
```
sudo apt-get install ros-kinetic-sick-tim
```

wizard007, 2018-02-14:
hi @robotpilot ,
thank you for replying even it is not related to tb3 issue. I have installed the sick-tim as to your advise and the cMake issue has been resolved.

robotpilot, 2018-02-14:
@wizard007 
You're welcome. Thank you and please ask if you have further questions. :)

closed 2018-02-14 by robotpilot