ROBOTIS-GIT/turtlebot3 #130: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

mw-sandipk, 2018-02-16:
turtlebot3@turtlebot3:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch 
... logging to /home/turtlebot3/.ros/log/50d0066c-1356-11e8-861b-a0c589238c19/roslaunch-turtlebot3-2515.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.31.28.61:44909/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

auto-starting new master
process[master]: started with pid [2533]
ROS_MASTER_URI=http://172.31.28.61:11311

setting /run_id to 50d0066c-1356-11e8-861b-a0c589238c19
process[rosout-1]: started with pid [2546]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [2549]
process[turtlebot3_lds-3]: started with pid [2550]
process[turtlebot3_diagnostics-4]: started with pid [2551]
[INFO] [1518812209.978335]: ROS Serial Python Node
[INFO] [1518812209.993142]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1518812227.101711]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812242.104960]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812257.107953]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812272.109768]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812287.111436]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812302.114523]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812317.117827]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812332.119702]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812347.121911]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812362.124240]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812377.127180]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812392.129857]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812407.131875]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812422.133763]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812437.136049]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812452.138707]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812467.141226]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812482.143648]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812497.145154]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812512.147175]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812527.149109]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812542.152116]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812557.155076]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812572.157114]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[ERROR] [1518812587.160317]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

I made sure the rosserial package is updated:

turtlebot3@turtlebot3:~$ sudo apt-get install ros-kinetic-rosserial-python 
[sudo] password for turtlebot3: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
[... 18 lines cut ...]
0 upgraded, 0 newly installed, 0 to remove and 95 not upgraded.

turtlebot3@turtlebot3:~$ sudo apt-get install ros-kinetic-rosserial-arduino
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-kinetic-rosserial-arduino is already the newest version (0.7.7-0xenial-20180125-163152-0800).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb cifs-utils dmeventd dmraid dpkg-repack gir1.2-clutter-1.0 gir1.2-clutter-gst-3.0 gir1.2-cogl-1.0 gir1.2-coglpango-1.0 gir1.2-gtkclutter-1.0 gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 kde-config-screenlocker
  kinit kio kpackagelauncherqml kpackagetool5 kpartx kpartx-boot kwayland-data kwayland-integration kwin kwin-common kwin-data kwin-x11 libbonobo2-0 libbonobo2-common libdebian-installer4 libdevmapper-event1.02.1
  libdmraid1.0.0.rc16 libfam0 libgnome-2-0 libgnome2-common libgnomevfs2-0 libgnomevfs2-common libkdecorations2-5v5 libkdecorations2private5v5 libkf5activities5 libkf5archive5 libkf5attica5 libkf5auth-data libkf5auth5
  libkf5bookmarks-data libkf5bookmarks5 libkf5calendarevents5 libkf5codecs-data libkf5codecs5 libkf5completion-data libkf5completion5 libkf5config-bin libkf5config-data libkf5configcore5 libkf5configgui5 libkf5configwidgets-data
  libkf5configwidgets5 libkf5coreaddons-data libkf5coreaddons5 libkf5crash5 libkf5dbusaddons-bin libkf5dbusaddons-data libkf5dbusaddons5 libkf5declarative-data libkf5declarative5 libkf5globalaccel-bin libkf5globalaccel-data
  libkf5globalaccel5 libkf5globalaccelprivate5 libkf5gpgmepp5 libkf5guiaddons5 libkf5i18n-data libkf5i18n5 libkf5iconthemes-bin libkf5iconthemes-data libkf5iconthemes5 libkf5idletime5 libkf5itemviews-data libkf5itemviews5
  libkf5jobwidgets-data libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kdelibs4support-data libkf5kdelibs4support5 libkf5kdelibs4support5-bin libkf5kiocore5 libkf5kiofilewidgets5 libkf5kiontlm5 libkf5kiowidgets5
  libkf5newstuff-data libkf5newstuff5 libkf5notifications-data libkf5notifications5 libkf5package-data libkf5package5 libkf5parts-data libkf5parts-plugins libkf5parts5 libkf5plasma5 libkf5plasmaquick5 libkf5quickaddons5
  libkf5service-bin libkf5service-data libkf5service5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5 libkf5wallet-bin libkf5wallet-data libkf5wallet5
  libkf5waylandclient5 libkf5waylandserver5 libkf5widgetsaddons-data libkf5widgetsaddons5 libkf5windowsystem-data libkf5windowsystem5 libkf5xmlgui-bin libkf5xmlgui-data libkf5xmlgui5 libkscreenlocker5 libkwalletbackend5-5
[... cut ...]
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia qml-module-qtquick-controls qml-module-qtquick-dialogs qml-module-qtquick-privatewidgets rdate ros-kinetic-control-toolbox ros-kinetic-diff-drive-controller
  ros-kinetic-forward-command-controller ros-kinetic-gazebo-ros-control ros-kinetic-joint-limits-interface ros-kinetic-joint-state-controller ros-kinetic-position-controllers ros-kinetic-transmission-interface snap-confine
  sonnet-plugins thermald user-setup
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 95 not upgraded.

mw-sandipk, 2018-02-16:
I think it got lost in the errors:

I made sure the rosserial package is updated:

turtlebot3@turtlebot3:~$ sudo apt-get install ros-kinetic-rosserial-python
turtlebot3@turtlebot3:~$ sudo apt-get install ros-kinetic-rosserial-arduino

Both are up to date.

routiful, 2018-02-19:
Thank you for sharing updated packages :)

As well as, 'sudo apt-get update', sudo apt-get upgrade' commands make update all installed packages.

Thanks
Darby

mw-sandipk, 2018-02-23:
Well I tried the two:

```
sudo apt-get update
sudo apt-get upgrade
```
And then tried running the` turtlebot3_core` but it has the same problem

```

turtlebot3@turtlebot3:~$ roslaunch turtlebot3_bringup turtlebot3_core.launch 
... logging to /home/turtlebot3/.ros/log/c413415a-18e4-11e8-a0e4-a0c589238c19/roslaunch-turtlebot3-2411.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.31.28.61:45717/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

auto-starting new master
process[master]: started with pid [2423]
ROS_MASTER_URI=http://172.31.28.61:11311

setting /run_id to c413415a-18e4-11e8-a0e4-a0c589238c19
process[rosout-1]: started with pid [2436]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [2439]
[INFO] [1519423147.564850]: ROS Serial Python Node
[INFO] [1519423147.579496]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1519423164.688909]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
^C[turtlebot3_core-2] killing on exit
[INFO] [1519423168.528072]: Send tx stop request
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

routiful, 2018-03-05:
Hello :)

Did you solve this problem??
I found similar problem in [ROS answers](https://answers.ros.org/question/235188/unable-to-sync-with-device/)

I think you need to check USB Port before you started

Thanks 
Darby

routiful, 2018-03-09:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-09 by routiful

captsisko, 2019-01-14:
I have exactly the same problem.

drNoob13, 2019-01-30:
If you have that problem, you may want to check two things:

1. The USB connection between the boards.
2. Run OpenCR setup again: http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/

dvonhausen, 2019-12-21:
try putting "#define USE_USBCON" at the very top of your Arduino sketch.

dvonhausen, 2019-12-21:
That's Arduino code.

javidaslam, 2022-08-10:
SOLVED : Hey, I encountered the same error , it related to baud rate ! i changed baud rate to 57600 and it worked ,error solved !
 
rosrun rosserial_arduino serial_node.py /dev/ttyACM0 _baud:=57600