ROBOTIS-GIT/turtlebot3 #132: ssh connection to turtlebot3-waffle drops everytime after 10-15 mins

mw-sandipk, 2018-02-16:
I have been noticing this for some time now, that the wifi connection to the turtlebot3 drops after sometime 

I have gone on and updated based on the following: **https://discourse.ros.org/t/turtlebot3-software-and-firmware-update-and-waffle-pi/3729

but still, the network has issues, 
- the turtlebot3 is connected to a wifi
- from remote PC I usually am connected to the turtlebot3 using ssh
- this ssh connected drops after 10-15 mins everytime

The only solution for me is to reboot turtlebot3 and then reestablish the ssh connection.

Have you guys seen this behavior? Is there a way to diagnose this problem?

kijongGil, 2018-02-19:
Hi :)

Did you set up your Raspberry Pi Configuration?

https://www.raspberrypi.org/documentation/remote-access/ssh/

If you've already tried it, I recommend you try changing your remote PC or wifi.

mw-sandipk, 2018-02-19:
I actually have a intel-joule turtlebot3-waffle (i mentioned in the title). Can you suggest something with respect to intel-joule for waffle.

nyc2muc, 2018-02-19:
Assuming you are running ubuntu on joule...had the same issue - solved it: disable sleep, hibernating, and hybrid powersafe options - in addition: disable Gnome and lightdm

nyc2muc, 2018-02-19:
more detailed: https://discourse.ros.org/t/turtlebot3-joule-stalled/2739

mw-sandipk, 2018-02-23:
I still have to try the suggested fix, unable to get hold of the micro HDMI  -> HDMI cable.

routiful, 2018-03-20:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

mw-sandipk, 2018-04-11:
@routiful please re-open the issue, and look at my last comment.

routiful, 2018-04-12:
Hi :)

Sorry for inconvenience, Please try to reinstall ubuntu on Joule by following [wiki](http://emanual.robotis.com/docs/en/platform/turtlebot3/joule_setup/#joule-setup)

Although it has same problem, please contact to customer support in ROBOTIS (contactus2@robotis.com) to exchange it.

Thanks

mw-sandipk, 2018-04-12:
@routiful I did mention
----
Now the ssh connection is lost and the device is no longer on network

I reboot -> the network does not come up at all.

I try to connect a Monitor to IntelJoule and now the microHDMI to HDMI does not show anything on all the monitors I try.
----

even for reinstall the HDMI output is needed :)

routiful, 2018-04-13:
Please try to reinstall right BIOS version we pointed. 

Although it remains problem, contact to customer support in ROBOTIS (contactus2@robotis.com) to exchange it.

Thanks :)

closed 2018-04-20 by robotpilot