ROBOTIS-GIT/turtlebot3 #134: transformation the laser scan data with robot pose matlab !!!!

fereydoon031, 2018-02-19:
hello guys, I want to simulate the bag data that I got from ROS in MATLAB.  I separated the pose and range data from the bag file. and I wrote a script to plot the laser data and build a map (I assumed that my poses are correct). when I run my code, It works well, till the robot turns!. at that point, the scan data turns in the opposite direction! till the robot gets straight again. and again when the robot turns again I  experience the same problem till the robot gets straight. I don't know whats wrong!!...the transformation I used works well except when the robot turn!!

[image]

kijongGil, 2018-02-20:
Hi, :)
What do you use package to get the laser data?

fereydoon031, 2018-02-20:
@kijongGil 
Hi, Thank you for your reply.
I just run SLAM with ROS and used "rosbag record -a" for recording the data, and then used the bag data in MATLAB with the following code:
>>a=rosbag('bag file name')
>>b=select(a, 'Time',[a.StartTime a.EndTime], 'Topic', '/scan');
>>c=readMessages(b);
(and for odometry i used \odom)
P.s--> timing between odometry and scan data was not the same so I catched the odometry data every five step and ignored the rest..

kijongGil, 2018-02-20:
I think that the frame_id of the scan data in rosbag file is "base_scan".
Could you make new rosbag file?  
Change the frame_id from "base_scan" to "odom". 
Refer to the link below,
http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#turtlebot-follower-demo

routiful, 2018-03-20:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-20 by routiful