ROBOTIS-GIT/turtlebot3 #135: applications package

francoiswindels, 2018-02-20:
Hello,

i'm using a TB3 burger from an ubuntu 16.04 remote machine

i tried to install you applications package following the procedure you describe at the top of

http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#turtlebot-follower-demo

catkin_make runs to a point  (~97%) and then sends out error messages until it ends with

Invoking "make -j4 -l4" failed

details of the error log attached

any thought on what went wrong?

thanks for your time and patience

François
[TB3_applications.pdf](https://github.com/ROBOTIS-GIT/turtlebot3/files/1738816/TB3_applications.pdf)

routiful, 2018-02-20:
Hello @francoiswindels  :)

Did you try download turtlebot3_msgs??

francoiswindels, 2018-02-20:
Hello,
yes i did
 git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
and then  git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
followed by cd ~/catkin_ws and catkin_make

cheers
F

routiful, 2018-02-22:
Hello :)

Delete ``build`` and ``devel`` folder in catkin_ws and try once again catkin_make command.

francoiswindels, 2018-02-23:
Hi,
thanks for following up,
i tried a build after deleting build and devel but that did not help
the problem seems to be coming from gazebo_ros_turtlebot3.cpp
any thought?

thanks
F

[error_log.docx](https://github.com/ROBOTIS-GIT/turtlebot3/files/1751467/error_log.docx)

PandaJules, 2018-02-25:
@francoiswindels 
So your error log is actually quite explanatory - it tells you:
"error: ‘robot_model’ was not declared in this scope"

That cpp file is written in a way, so that it works on both Burger and Waffle without duplicating code. When you run it, it looks for a variable robot_model and from then onwards, it knows whether you want to be working with Burger or Waffle. So all you need to do is initialize it. If you keep working with different models, you can initialize it in the terminal every time you decide to do some coding (and then when you close the terminal, your variable will disappear). Or (and that is what most people do, because who has money and time for both versions 😋), you just add an extra line to .bashrc file. It is actually described in the official manual here: http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/

Just go 
` gedit ~/.bashrc`
add line export `TURTLEBOT3_MODEL = ` ... (whatever you have)
`source ~/.bashrc`

francoiswindels, 2018-02-27:
Hi,
thanks for taking time to look into this problem,
i do have the turtlebot 3 model specified in my bashrc,
i sourced bashrc before catkin_make but
the problem persist.

thanks for your patience
F

PandaJules, 2018-02-27:
Okay, but if you look at your log, all errors start at line 45
If I am not mistaken, this is where it crashes:

`bool GazeboRosTurtleBot3::init()
{
  // initialize ROS parameter
  nh_.param("is_debug", is_debug_, is_debug_);
  std::string robot_model = nh_.param<std::string>("tb3_model", "");

  if (!robot_model.compare("burger"))
  {
    turning_radius_ = 0.08;
    rotate_angle_ = 50.0 * DEG2RAD;
    front_distance_limit_ = 0.7;
    side_distance_limit_  = 0.4;
  }`

So just 2 lines above `robot_model.compare("burger")` we have both declaration and initialization of the variable robot_model. You could put a print statement after it to see whether `nh_.param<std::string>("tb3_model", "")` was actually able to find it out. 

Then your log says "turning_radius_’, ‘rotate_angle_’ , ‘front_distance_limit_’ , ‘side_distance_limit_’ was not declared in this scope". What you are doing inside "if" is initializing, but those variable should have been declared in a header file (gazebo_ros_turtlebot3.h which is inside "include" folder)

francoiswindels, 2018-02-28:
HI,
so you agree with me that the deluge of error message is triggered by the
build of the gazebo package for turtlebot3?
maybe i should start with
1/delete that package
2/re run catkin_make on my catkin_ws to make sure i get a clean build
3/re-install the gazebo turtlebot3 package (from
https://github.com/ROBOTIS-GIT/turtlebot3_simulations)
4/build that package only by using catkin_make -D
CATKIN_WHITELIST_PACKAGES="turtlebot3_simulations"
what do you think?

thanks for your time and patience
cheers
F

francoiswindels, 2018-03-02:
Hi,
 i tried just that and it all worked.
thanks
F

closed 2018-03-02 by francoiswindels