ROBOTIS-GIT/turtlebot3 #137: burger SLAM

mathgameapi, 2018-02-27:
My step:

[Remote PC] - roscore

[TURTLEBOT] -  roslaunch turtlebot3_bringup turtlebot3_robot.launch

[Remote PC] - export TURTLEBOT3_MODEL=burger

[Remote PC] - roslaunch turtlebot3_slam turtlebot3_slam.launch

[Remote PC] - rosrun rviz rviz -d `rospack find turtlebot3_slam`/rviz/turtlebot3_slam.rviz

[Remote PC] - roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch

After I launch the rviz, I got some data and laserscan status is ok.
After i launch the teleop_key, i try to control the robot to move. the lasterscan status will become error and rviz never update the screen.

routiful, 2018-02-27:
Hello :)

May you show me your error message in terminal and your rviz window??
And I recommend to synchronize TB3 and RemotePC
This link will help you

[Timesync](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remotepc
)
Thanks
Darby

mathgameapi, 2018-03-01:
@routiful  Any update? Thanks

routiful, 2018-03-05:
Hello :)

Don't you upgrade your OpenGl version to latest version?
I used to version 4.5 (GLSL 4.5)

mathgameapi, 2018-03-05:
@routiful 
My opengl information as follow:
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Haswell Mobile 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 17.2.8
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 17.2.8
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 17.2.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

As that config , we cannot run the slam code?

routiful, 2018-03-05:
No, but i assume that Rviz in your laptop has problem because of error messages(XmbTextListToTextProperty result code -2) and opengl version.

If not, could you show me rviz window when you open slam package?

Thanks
Darby

mathgameapi, 2018-03-06:
@routiful 

The log and screen capture as folllow : Thanks

export TURTLEBOT3_MODEL=burger
roslaunch turtlebot3_slam turtlebot3_slam.launch 

... logging to /home/astr
... logging to /home/astri/.ros/log/a7617bb6-20d8-11e8-a33a-28d2449a742a/roslaunch-astri-ThinkPad-Edge-E540-2672.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://192.168.11.2:45740/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 50.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /turtlebot3_slam_gmapping/angularUpdate: 0.2
 * /turtlebot3_slam_gmapping/astep: 0.05
 * /turtlebot3_slam_gmapping/base_frame: base_footprint
 * /turtlebot3_slam_gmapping/delta: 0.05
 * /turtlebot3_slam_gmapping/iterations: 5
 * /turtlebot3_slam_gmapping/kernelSize: 1
 * /turtlebot3_slam_gmapping/lasamplerange: 0.005
 * /turtlebot3_slam_gmapping/lasamplestep: 0.005
 * /turtlebot3_slam_gmapping/linearUpdate: 0.2
 * /turtlebot3_slam_gmapping/llsamplerange: 0.01
 * /turtlebot3_slam_gmapping/llsamplestep: 0.01
 * /turtlebot3_slam_gmapping/lsigma: 0.075
 * /turtlebot3_slam_gmapping/lskip: 0
 * /turtlebot3_slam_gmapping/lstep: 0.05
 * /turtlebot3_slam_gmapping/map_update_interval: 2.0
 * /turtlebot3_slam_gmapping/maxUrange: 4.0
 * /turtlebot3_slam_gmapping/minimumScore: 100
 * /turtlebot3_slam_gmapping/odom_frame: odom
 * /turtlebot3_slam_gmapping/ogain: 3.0
 * /turtlebot3_slam_gmapping/particles: 120
 * /turtlebot3_slam_gmapping/resampleThreshold: 0.5
 * /turtlebot3_slam_gmapping/sigma: 0.05
 * /turtlebot3_slam_gmapping/srr: 0.01
 * /turtlebot3_slam_gmapping/srt: 0.02
 * /turtlebot3_slam_gmapping/str: 0.01
 * /turtlebot3_slam_gmapping/stt: 0.02
 * /turtlebot3_slam_gmapping/temporalUpdate: 0.5
 * /turtlebot3_slam_gmapping/xmax: 10.0
 * /turtlebot3_slam_gmapping/xmin: -10.0
 * /turtlebot3_slam_gmapping/ymax: 10.0
 * /turtlebot3_slam_gmapping/ymin: -10.0

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    turtlebot3_slam_gmapping (gmapping/slam_gmapping)

ROS_MASTER_URI=http://192.168.11.2:11311

process[robot_state_publisher-1]: started with pid [2692]
process[turtlebot3_slam_gmapping-2]: started with pid [2693]
[ INFO] [1520297636.019513120]: Laser is mounted upwards.
 -maxUrange 4 -maxUrange 3.49 -sigma     0.05 -kernelSize 1 -lstep 0.05 -lobsGain 3 -astep 0.05
 -srr 0.01 -srt 0.02 -str 0.01 -stt 0.02
 -linearUpdate 0.2 -angularUpdate 0.2 -resampleThreshold 0.5
 -xmin -10 -xmax 10 -ymin -10 -ymax 10 -delta 0.05 -particles 120
[ INFO] [1520297636.021760563]: Initialization complete
update frame 0
update ld=0 ad=0
Laser Pose= -0.032 1.6018e-05 3.14109
m_count 0
Registering First Scan
update frame 3
update ld=5.47314e-05 ad=0.000401333
Laser Pose= -0.032 1.25998e-05 3.1412
m_count 1
Average Scan Matching Score=224.133
neff= 120
Registering Scans:Done
update frame 6
update ld=4.71974e-05 ad=0.000424529
Laser Pose= -0.032 1.71387e-05 3.14106
m_count 2
Average Scan Matching Score=228.228
neff= 120
Registering Scans:Done
update frame 9
update ld=3.68393e-05 ad=0.000155283
Laser Pose= -0.032 1.51831e-05 3.14112
m_count 3
Average Scan Matching Score=225.277
neff= 120
Registering Scans:Done
update frame 12
update ld=1.43604e-05 ad=0.000270015
Laser Pose= -0.032 1.5161e-05 3.14112
m_count 4
Average Scan Matching Score=225.14
neff= 120
Registering Scans:Done
update frame 15
update ld=2.27025e-05 ad=0.000149577
Laser Pose= -0.032 1.80025e-05 3.14103
m_count 5
Average Scan Matching Score=231.293
neff= 120
Registering Scans:Done
update frame 18
update ld=3.70687e-05 ad=0.000261456
Laser Pose= -0.0320018 2.05776e-05 3.14095
m_count 6
Average Scan Matching Score=234.343
neff= 120
Registering Scans:Done
update frame 21
update ld=1.00469e-05 ad=0.000271238
Laser Pose= -0.032 1.49045e-05 3.14113
m_count 7
Average Scan Matching Score=231.71
neff= 120
Registering Scans:Done
update frame 24
update ld=3.61469e-05 ad=0.000236085
Laser Pose= -0.032 1.50327e-05 3.14112
m_count 8
Average Scan Matching Score=231.78
neff= 120
Registering Scans:Done
update frame 27
update ld=1.3028e-05 ad=0.000393201
Laser Pose= -0.032 1.30763e-05 3.14118

[image]

routiful, 2018-03-09:
It happens because of tf time error. 
We highly recommend TB3 and RemotePC is connected under internet due to timesync.

routiful, 2018-03-20:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-20 by routiful