ROBOTIS-GIT/turtlebot3 #140: Motor Setup Issue

ludusrusso, 2018-03-06:
Hey! I've replaced a Dynamixel for my turtlebot.. 
Strangely, it is not recognized by the robot, even if checked the setup: 

 - firmware 41
 - ID 1
 - Baud 1Mb

Anyway, when I try to run the "setup motor" sketch, I got this error:

```

1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
>> Do you really want to setup ? y/n : yes
setup.... left
Find Motor...
    setbaud : 57600

This function is deprecated. Use 'Serial.print()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
    setbaud : 1000000

This function is deprecated. Use 'Serial.print()' and 'getRxPacketError()' instead
[TxRxResult] There is no status packet!
 
 
1. setup left  motor
2. setup right motor
3. test  left  motor
4. test  right motor
```

Any help?

routiful, 2018-03-06:
Hello :)

You need to check

First, you have to power on to the Dynamixel
Second, OpenCR is connected with single Dynamixel

Thanks
Darby

ludusrusso, 2018-03-07:
Hi :)
I tested both one or two dynamixel at the same time.. 

In my understanding, the dynamixel powered by the OpenCR board right? I've connected it with the turtlebot3 battery and the red led on the dynamixel blinks when I turn it on! So it should be powered on.

routiful, 2018-03-08:
Then it might be baudrate of yours is not 57600 or 1000000.

I recommend to use monitor.ino in dynamixel-workbench example for check baudrate.
(File -> Examples -> OpenCR -> DynamixelWorkbench -> Monitor)

Thanks
Darby

routiful, 2018-03-20:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-20 by routiful