ROBOTIS-GIT/turtlebot3 #142: Turtlebot 3 bugger stop moving after update firmware

fllay, 2018-03-11:
It works fine until I updated the firmware in openCR. Motors dont move anymore. Lidar is ok. Very frustrate. Please help. All the motor test programs fail as well. it said not respond from motor. It is not a connection problem since It work. Is before with previous firmware. I update it as the system told me to check openCR firmware.

routiful, 2018-03-11:
Hello :)

Can you show me what happen to your TB3 as Rviz, Terminal and Which firmware version do you use?

fllay, 2018-03-12:
Rviz in SLAM mode is fine but I did not try Rviz in navigation mode since the motor does not move.  The map appears on Rviz but I could not move the robot using keyboard
 teleop. The firmware on OpenCR board is version 1.0.17. I used the old firmware before it is just fine but there are a message  about let me check the firmware version on the OpenCR board when I run bring up script.  I have also tried to use motor test but it does not work. It complains about the motor does not reply for both left and right motor. As I mention It works ok before until I update the firmware on OpenCR board.

routiful, 2018-03-13:
Hi :)

Firmware version 1.1.0 has bug to control dynamixel.
We hotfixed it and now you can download v.1.1.1.
We apologize for any inconvenience.

Thanks
Darby

fllay, 2018-03-15:
It's moving now. Thank you very much.

closed 2018-03-20 by routiful