ROBOTIS-GIT/turtlebot3 #149: Warning on bringup.

shahvi, 2018-03-12:
HI, I have a brand new TB3 and have followed all the instructions from the manual.  I have also updated the OpenCR using the shell script in the manual.  However, when I run `roslaunch turtlebot3_bringup turtlebot3_robot.launch`, the following warning shows up. 

> [ WARN] [1520881835.413400698]: Check OpenCR update and change your firmware!!

Everything on the bot seems to work just fine, however it is annoying to see these warn messages.  Also, I am concerned that it will fillup the log files quickly.

Any ideas what might be going on here?

kijongGil, 2018-03-16:
Hi :)

Your OpenCR is not updated. If you failed to update OpenCR, try writing bootloader. 
 
http://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-ide
6.1.7 Writing Bootloader(Linux)

Thank you.

Shahnama, 2018-03-16:
Hi, I have the same problem. OpenRC seems is updated to version 1.1.2, It can connect to OpenCR recognize version (1.1.2) and can Start Calibration of Gyro and finish it but after that, it says checkOpenCR update and change your firmware!

routiful, 2018-03-16:
Hi :)

Have you downloaded the latest version of software?

Shahnama, 2018-03-16:
I have used Linux based on Raspbian so I have considered I do not need to update any software! Becuase for ubuntu in the wiki it asked for an update but for Raspbian based I haven't seen such structure.

routiful, 2018-03-16:
I mean the package for turtlebot3 in github not OS

Shahnama, 2018-03-16:
I have done following, is it correct?

$ cd ~/catkin_ws/src
$ sudo rm -rf turtlebot3/
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3.git

$ cd ~/catkin_ws/src
$ sudo rm -rf turtlebot3_msgs/
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git

routiful, 2018-03-16:
Yes, correct :) 

Do you still see the [WARN] message?

Shahnama, 2018-03-16:
Yes, I can see it

kingnavid, 2018-03-16:
Any ideas

Shahnama, 2018-03-16:
any ideas?

shahvi, 2018-03-16:
@kijongGil I have updated the bootloader using the arduino IDE to 1.1.2 and then also then updated the firmware using the shell script.  I still see the WARN messages.

mirwox, 2018-03-18:
I'm also having the same error message with everything up to date  (1.1.2 firmware).

I noticed that if I bringup the core and lidar separately, there error message does not come up.

roslaunch turtlebot3_bringup turtlebot3_lidar.launch
roslaunch turtlebot3_bringup turtlebot3_core.launch

kijongGil, 2018-03-19:
We tested bringup on Ubuntu and Raspbian, working well.

please remove the turtlebot3 package, and then install and build again.(catkin_make)

shahvi, 2018-03-19:
I observed the same behavior as @mirwox .  However, after I pull the code from git again, the problem seems to have gone away.
I will close this for now.  Thanks for your help everyone.

closed 2018-03-19 by shahvi

turtlezs, 2018-05-02:
Hi @shahvi,

Would you mind please sharing the exact steps you have taken to fix the error. 

I have done the following but it did not solve the problem for me:

$ cd ~/catkin_ws/src
$ sudo rm -rf turtlebot3/
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3.git

$ cd ~/catkin_ws/src
$ sudo rm -rf turtlebot3_msgs/
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git

It will be greatly appreciated. 

Thanks in advance.

vik748, 2018-05-02:
@turtlezs make sure that opencr is updated. Additionally try removing the Dev and build folders from the workspace so that everything has to be built again.

settonfarms, 2018-05-02:
Hi vik748,

Could you list steps?

Thank you,

vik748, 2018-05-03:
1. Update the OpenCR boot loader (ver 1.1.2) using the Arduino IDE http://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-ide 6.1.7 Writing Bootloader(Linux)
2. Update the firmware using linux scipt http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr-setup
3. Update packages and make 
`$ cd ~/catkin_ws/
$ rm build/ dev/
$ cd src
$ rm -rf turtlebot3/ turtlebot3_msgs/
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
$ git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git`
$ cd ~/catkin_ws
$ catkin_make
$ source devel/setup.bash

Hope that helps...

turtlezs, 2018-05-03:
Thank you @vik748!

For the first step “Update the OpenCR boot loader“, do you install “Arduino IDE” on the Raspberry or do you need to install it on another PC and unplug the OpenCR board from the Raspberry and plug it into the PC?

shahvi, 2018-05-03:
I installed it on a PC.

settonfarms, 2018-05-04:
@vik748 
Thank you

turtlezs, 2018-05-08:
@vik748 <https://github.com/vik748>

It worked.

Thank you!