ROBOTIS-GIT/turtlebot3 #151: motors getting locked with using with intel joule and openCR (turtlebot3)

allanis79, 2018-03-13:
Hello all,

I recently bought turtleobt3, it came with dynamixel motors Xm430-w210-t, intel joule and openCR board. I followed the tutorial and done all the steps. 

when i launched turtlebot3 bringup and lauched teleoperation it didn't show any errors in the terminal, but the turtlebot didn't move at all when i pressed the keys. I can see the output on /odom topic. 

It looks like motors got locked. I tired rotating them manually but they didn't rotate. I didn't force, i thought i might break something inside.

Can anyone help me what's happening?

Thanks

routiful, 2018-03-13:
Hi :)

Firmware version 1.1.0 has bug to control dynamixel.
We hotfixed it and now you can download v.1.1.1.
We apologize for any inconvenience.

Thanks
Darby

allanis79, 2018-03-13:
where and how should i download it?

OpusK, 2018-03-14:
Hello, @allanis79 

You can download it using arduino or shell.
Please refer to [e-manual](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr10-firmware-upload-for-tb3) for details.

allanis79, 2018-03-14:
thanks, I will try it tomorrow. @OpusK

routiful, 2018-03-20:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever.
Thanks. :)

closed 2018-03-20 by routiful