ROBOTIS-GIT/turtlebot3 #152: Navigation

mathgameapi, 2018-03-14:
I forward the guide to create the map and load the map
I don't know why i open the rviz will got the follow result and cannot control the robotis

[image]
Make the map screen

[image]
Load the map screen

routiful, 2018-03-15:
Hi :)

The map you created is too small to drive TB3.
I recommend you would check it chapter 11.7.1 Costmap in [ROS Robot Programming](https://community.robotsource.org/t/download-the-ros-robot-programming-book-for-free/51/23)

Best regards
Darby

mathgameapi, 2018-03-15:
@routiful 

When i run the navigation sample, if the map is small will trigger the map rotation problem?

routiful, 2018-03-20:
Hi :)

Sorry for late reply.
The map rotation problem is caused by failed localization.
You can solve it by making bigger map

mathgameapi, 2018-04-03:
@routiful 

I made the map to double size still have the rotation problem.

What is the size you prefer for the map?

routiful, 2018-04-03:
Please show me how it works by capture your monitor

mathgameapi, 2018-04-04:
@routiful  
would you like to tell me how can i give you the video recording? (over 200MB)
Thanks

mathgameapi, 2018-04-10:
@routiful 
Still not work

How can i send you the video ?

kijongGil, 2018-04-11:
@mathgameapi 
Would you like to upload capture you worked?

If you want to send your video, upload your video on youtube and tell me link.

mathgameapi, 2018-04-12:
@kijongGil 

Video 1 :http://megashare.megahd.com.br/47e605e1c6204e39ae253837f04e2cb7 

Video 2 :
http://megashare.megahd.com.br/04666f9d560ed49cf5b23615416093c2 
http://megashare.megahd.com.br/2fc3558b70605755c67e7fc050fdedfe 
http://megashare.megahd.com.br/750a1aa57765b3d6069c1b60194f372a 
http://megashare.megahd.com.br/9ae81fb1d1acdde365b1d43f04890573

kijongGil, 2018-04-12:
I can't watch your video 2, It can't extract.

Your map looks troubled in video 1.

please remake a map with slam package. Then, place the TB3 in the same position when you start slam and navigation. 

If the problem isn't solved, show me the error messages in your terminal.

mathgameapi, 2018-04-12:
@kijongGil 
[image]
[image]
[image]
[image]

kijongGil, 2018-04-12:
Could you move the TB3 out of the area surrounded by red lines? 

Because the meaning of the red line is the area where robots can not go. But, your TB3 is located in this area.

robotpilot, 2018-04-18:
I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue. Also, you can reopen this issue to show this issue to the users whenever. Thanks. :)

closed 2018-04-18 by robotpilot