ROBOTIS-GIT/turtlebot3 #157: turtlebot3 build map with cartographer

ncnynl, 2018-03-18:
I have tested turtlebot3 burger to use cartographer  , see : []( https://github.com/ncnynl/turbot3)

routiful, 2018-03-20:
Thanks you for your test :)

ncnynl, 2018-03-21:
Now  have supported  building map with cartographer , hector,  karto   :)

robotpilot, 2018-05-17:
Hi @ncnynl

Thank you for sharing your idea and code. I want to thank you for your contribution. I did several tests as you guided it and reflected it in our develop branch. I will release it in the next TurtleBot3 software release.
https://github.com/ROBOTIS-GIT/turtlebot3/tree/develop/turtlebot3_slam/launch

Thank you again.

ncnynl, 2018-05-18:
@robotpilot   I have checked your code . good job.

robotpilot, 2018-05-20:
Thank you very much. We will also apply it to the actual robot and release it this week.

robotpilot, 2018-05-25:
Hi, @ncnynl !

We update and release new firmware (version 1.2.0) and new software (version 1.0.0) for users of TurtleBot3. :) 

Thank you to your contribution!

Please refer to the link below for details of this update.

https://discourse.ros.org/t/announcing-turtlebot3-software-v1-0-0-and-firmware-v1-2-0-update/4888

closed 2018-05-25 by robotpilot