ROBOTIS-GIT/turtlebot3 #158: TB3 burger cheksum error message

francoiswindels, 2018-03-18:
Hello,
my turtlebot 3 burger stopped working after a software upgrade (apt-get update && apt-get upgrade)
when i try roslaunch turtlebot3_bringup turtlebot3_robot.launch 
i get the usual startup message and then
 Connecting to /dev/ttyACM0 at 115200 baud
Creation of publisher failed: Checksum does not match: d537ed7b8d95065b6c83830430b93911,427f77f85da38bc1aa3f65ffb673c94c
 Note: publish buffer size is 1024 bytes

full log attached
my TB3 is running ubuntu mate 16.04
can you help me with that?

[TB3_error_message.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/1822428/TB3_error_message.txt)

routiful, 2018-03-18:
Hi :)

Have you uploaded firmware version(1.1.2)?
This link will help

[Firmware upload for tb3](http://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/#opencr10-firmware-upload-for-tb3)

Thanks
Darby

francoiswindels, 2018-03-19:
Thanks Darby,
that fixed it, nice and easy
cheers
F

closed 2018-03-19 by francoiswindels