ROBOTIS-GIT/turtlebot3 #159: Make Turtlebot Bringup Works with Kinect v2

rui-cao, 2018-03-18:
Hi,

I would like to make a Turtlebot simulation works with Kinect V2. I was following this tutorial (http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebothttp://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot). 

I did the following attempt:

Kinect2 works with the driver freenect2 and can be communicated with ROS using the package iai_kinect2/kinect2_bridge.

I modified the turtlebot_bringup/launch/includes/3dsensor/kinect.launch.xml by chaning the driver include file to launch file for kinect2_bridge, and delete the arguments that are not used by kinect2_bridge. Here is the modified 3dsensor/kinect.launch.xml:

```xml
<launch>
  <!-- "camera" should uniquely identify the device. All topics are pushed down
       into the "camera" namespace, and it is prepended to tf frame ids. -->
  <arg name="publish_tf"  default="false"/>

  <!-- Processing Modules -->                                    
                                                                                                                                                      
  <!-- <include file="$(find freenect_launch)/launch/freenect.launch">  --> 
<include file="/home/user_name/catkin_ws/src/iai_kinect2/kinect2_bridge/launch/kinect2_bridge.launch">                           
    <arg name="publish_tf"                      value="$(arg publish_tf)"/>                                          
                                                                                                                
    <!-- Processing Modules -->                                                                         
  </include>                                                                                        
</launch>
```
However, when I call up the turtlebot_bringup, I get this error:
```
unused args [num_worker_threads, rgb_processing, depth_registration, camera, depth_processing, 
ir_processing, disparity_processing, disparity_registered_processing, depth_registered_processing]
 for include of [/opt/ros/kinetic/share/turtlebot_bringup/launch/includes/3dsensor/kinect.launch.xml]
```

I don't understand this error, since I already deleted those arguments that are not used by the kinect2_bridge.launch, but I still get the above error.

Does anyone knows how to solve this problem?

Please let me know if I did not provide enough infomation!

routiful, 2018-03-19:
Hi :)

This repository supports only for TurtleBot3 which is different turtlebot.
I think you would take more information about your problem in [turtlebot_repo](https://github.com/turtlebot/turtlebot) or ROS [answer](https://answers.ros.org/questions/).

Thanks 
Darby

routiful, 2018-03-20:
If anyone want to trace this issue, please following this link [ROS Answer](https://answers.ros.org/question/285845/make-turtlebot-bringup-works-with-kinect-v2/)

closed 2018-03-20 by routiful