ROBOTIS-GIT/turtlebot3 #160: Rosserial Publisher Error

hiseholuwa, 2018-03-21:
I encountered the following error on running the turtlebot_core.launch file on the turtlebot. 
```
hiseholuwa@hiseholuwa:~/test_ws$ roslaunch turtlebot3_bringup turtlebot3_core.launch
... logging to /home/hiseholuwa/.ros/log/5af139f2-2c9e-11e8-b605-94659c20af6c/roslaunch-hiseholuwa-20150.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:46547/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)

auto-starting new master
process[master]: started with pid [20160]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5af139f2-2c9e-11e8-b605-94659c20af6c
process[rosout-1]: started with pid [20176]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [20183]
[INFO] [1521591928.457233]: ROS Serial Python Node
[INFO] [1521591928.462275]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1521591930.571028]: Creation of publisher failed: unpack requires a string argument of length 2
[ERROR] [1521591945.569000]: Lost sync with device, restarting...
[ERROR] [1521591945.572016]: Creation of publisher failed: unpack requires a string argument of length 2
^C[turtlebot3_core-2] killing on exit
[INFO] [1521591951.924091]: Send tx stop request
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
I encounter the same error when I run the tuttlebot3_robot.launch file.
```
hiseholuwa@hiseholuwa:~/test_ws$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
... logging to /home/hiseholuwa/.ros/log/e4bcfb7c-2ca2-11e8-b605-94659c20af6c/roslaunch-hiseholuwa-24517.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:45147/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

auto-starting new master
process[master]: started with pid [24527]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e4bcfb7c-2ca2-11e8-b605-94659c20af6c
process[rosout-1]: started with pid [24540]
started core service [/rosout]
process[turtlebot3_core-2]: started with pid [24543]
process[turtlebot3_lds-3]: started with pid [24554]
process[turtlebot3_diagnostics-4]: started with pid [24559]
[INFO] [1521593877.708350]: ROS Serial Python Node
[INFO] [1521593877.715629]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1521593879.823935]: Creation of publisher failed: unpack requires a string argument of length 2
^C[turtlebot3_diagnostics-4] killing on exit
[turtlebot3_lds-3] killing on exit
[turtlebot3_core-2] killing on exit
[INFO] [1521593882.998554]: Send tx stop request
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
I have updated my rosserial package and also made sure I have the latest firmware on my OpenCR which is version 1.1.2, I believe. Any and all help on this will be highly appreciated. Thanks.

hiseholuwa, 2018-03-21:
Update: So I uninstalled the OpenCR board from Arduino and reinstalled it. I also reflashed the OpenCR bootloader and flash the turtlebot3_core program to it. When I run the launch file it works properly now, both for the turtlebot3_core.launch and turtlebot3_robot.launch. But this is when I changed the serial_node.py port to /dev/ttyACM1 and not /dev/ttyACM0 that comes with the file. Is there a reason for this?? Thanks.

kijongGil, 2018-03-22:
Hi :)

Did you connect usb port to some device?

hiseholuwa, 2018-03-26:
No, only the Lidar and OpenCR were connected to the Pi.

hiseholuwa, 2018-03-28:
Update - I re-installed the ubuntu mate system and re-installed ROS on the Pi and everything is working fine now. Got no clue as to what went wrong the first time. Thanks.

closed 2018-03-28 by hiseholuwa

irsyadtc, 2019-07-25:
> [INFO] [1521650797.868662]: Calibration End
> [ERROR] [1521650797.881179]: Message from device dropped: message larger than buffer.
> [ERROR] [1521650797.920229]: Message from device dropped: message larger than buffer.
> [ERROR] [1521650797.960913]: Message from device dropped: message larger than buffer.
> [ERROR] [1521650798.001095]: Message from device dropped: message larger than buffer.
> [ERROR] [1521650798.040901]: Message from device dropped: message larger than buffer.
> [ERROR] [1521650798.080813]: Message from device dropped: message larger than buffer.
> [ERROR] [1521650358.853689]: Mismatched protocol version in packet: lost sync or rosserial_python is from different ros release than the rosserial client
> [INFO] [1521650358.854188]: Protocol version of client is Rev 0 (rosserial 0.4 and earlier), expected Rev 1 (rosserial 0.5+)
> ```
> 
> I ran a simple Arduino ROS program that utilizes rosserial on the OpenCR board and it ran perfectly fine when I connect to the /dev/ttyACM1 port and not the /dev/ttyACM0 though I get a TCP/IP connection failed warning after a while.

Regarding this error, I got the answer from 
[](https://answers.ros.org/question/73627/how-to-increase-rosserial-buffer-size/)

Edit the content in node_handle.h
`gedit ~/Arduino/libraries/ros_lib/ros/node_handle.h`

Under the /* Node Handle */:
I tried decrease MAX_SUBSCRIBERS to 15
MAX_PUBLISHERS = 15
increase the OUTPUT_SIZE to 1024

```
/* Node Handle */
template<class Hardware,
	 int MAX_SUBSCRIBERS = 15,
         int MAX_PUBLISHERS = 15,
         int INPUT_SIZE = 512,
         int OUTPUT_SIZE = 1024>

```
The INPUT_SIZE refers to subcribers buffer and OUTPUT_SIZE as publishers buffer.
Close and save the file. Then from Arduino IDE, upload the sketch into OpenCR.

Note: I made some changes in  turtlebot3_core sketch for my project, anyhow others can try a different value.

ROBOTIS-Kayman, 2019-07-25:
@irsyadtc 
Thank you for your contribution.

Regards,
Kayman