ROBOTIS-GIT/turtlebot3 #162: OpenCR can't read motor encoder

ludusrusso, 2018-03-29:
Hi,
my turtlebot works great, but it can't read the motor encorders, while is able to control them very well.

Here the motor info:
 - firmware version 41
 - protocol 2.0

The topic `/sensor_state` is never published, while topic `joint_states` return the same value:

```
header: 
  seq: 759
  stamp: 
    secs: 1522323390
    nsecs: 940208339
  frame_id: base_link
name: ['wheel_left_joint', 'wheel_right_joint']
position: [0.0, 0.0]
velocity: [0.0, 0.0]
effort: [7.335467576660691e+21, 143.15293884277344]

```

routiful, 2018-03-30:
Hi :)

You need to check your setup. Please following below tutorial.
[setup dynamixel](http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#turtlebot3-setup-dynamixels)

ludusrusso, 2018-04-03:
HI,
I tried it, but the OpenCR is not able to detect any motor.. It seems that it's able to receive information from the motors, even if all works well with U2D2!

It may be an hardware issue?

routiful, 2018-04-04:
When you excute setup_motor, you have to connect only **single** Dynamixel to OpenCR. 
Please try it again, we will add this warning sentence in WIKI.

Sorry for inconvenience.

ludusrusso, 2018-04-04:
Hi!
It is very clear and I do connect only one motor! I’ve also tried others example like ping, but no one is able to detect any motor!

routiful, 2018-04-05:
Then it is caused by hardware faults.
You can take more information about exchange it.
Here is an email address.  contactus2@robotis.com

closed 2018-04-16 by routiful