ROBOTIS-GIT/turtlebot3 #163: Path planning failing

allanis79, 2018-03-29:
Hello,

I collected a map using SLAM and I am using that map for turtlebot3 navigation using AMCL. Whenever i click on goal pose, the path planning algorithm is planning through grey area. I am not sure why this is happening. Can someone help me? 

Thanks

allanis79, 2018-03-29:
@routiful can you help me?

routiful, 2018-03-30:
Hi :)

Can you show me your problem by capture image?

allanis79, 2018-03-30:
[image]

You can see the black line (path planned), it's planning through the grey area, why is this happening?

routiful, 2018-04-02:
Before you set goal pose by _2D Goal Nav_, you have to set initial pose of TB3 by _2D Pose Estimate_.

As shown in your capture, the scan data is not correctly on map. 
This is the reason why your TB3 is not working.

allanis79, 2018-04-03:
I did set the initial position using 2D pose estimate. I think what you mentioned is not the reason. In first case, path planning algorithm shouldn't generate a path through grey area. 

This github repo is really broken. Lot's of things doesn't work properly.

routiful, 2018-04-03:
Please show me how it works by capture your monitor.

I don't understand what you said 'broken'. This repo have been using many people in the world. 
Because of different development environment, a lot of problems are caused in any packages.

We are sorry for inconvenience. but we always ready for support.
If you have any problem to this repo you said 'broken', please feel free to open issue

Thanks a lot

allanis79, 2018-04-05:
Hello,

I replaced all the parameters in TB3 with TB2 and it worked. I am not exactly sure what parameter is making the difference. 

Maybe this package worked for lot of people, but i wasn't able to get it running.

I am using a modified version of this file [https://github.com/turtlebot/turtlebot_apps/blob/indigo/turtlebot_navigation/launch/amcl_demo.launch](url)

routiful, 2018-04-19:
Thank you for sharing. I will review that parameters and engage it to improve our packages.

closed 2018-04-19 by routiful